FreeOpcUa/python-opcua #616: server error

akgaurihar, 2018-06-06:
Hello People,
I am trying to create opc ua server on my windowns, but I am getting this kind of error while running the server. Somebody please help me to clear this error.

"add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists".

For your information I am attaching a code with this post.
[server code.txt](https://github.com/FreeOpcUa/python-opcua/files/2076681/server.code.txt)

oroulet, 2018-06-06:
This are warning from standard address space. Nothing to care about. They are removed now

akgaurihar, 2018-06-06:
The output doesn`t show the values of the parameters mentioned in the code .
 It only says ,"Server started at opc.tcp://192.168.113.195:8080"

closed 2018-06-07 by akgaurihar