FreeOpcUa/python-opcua #617: Some nodes from standard_address_space_part9 not serializable

Nebukadneza, 2018-06-07:
While walking through nodes of the standard address space, part 9, I have discovered some nodes that are broken: Some of their attributes cannot be serialized to binary using `opcua.ua.ua_binary.to_binary(…)`. The nodes affected are all nodes which have a `LocalizedText` as a `Value` attribute. This attribute gets written by `schemas/generate_address_space.py` as follows:
```
attrs.Value = ua.Variant([('Locale', 'en'), ('Text', '<The Text>')], ua.VariantType.LocalizedText)
```
When trying to serialize this Variant (embedded in a `DataValue`), the unpacking-and-serializing-stack eventually ends up in `struct_to_binary`, which will attempt to `for name, uatype in "Locale".ua_types` — but of course `str` doesn’t have a `.ua_types`.

When changing the code-generator by adding the special-case of `LocalizedText` as a Value, thus generating this type of code:
```
attrs.Value = ua.Variant(LocalizedText("<The Text>"), ua.VariantType.LocalizedText)
```
Then serialization works as expected.

The diff on `schemas/generate_address_space.py` is rather small, but before PR’ing anything, I wanted to ask if i understood the problem correctly, and this is the right way of solving it.

Thank & Best Regards
-Dario

oroulet, 2018-06-07:
Can you create the PR anyway? It is easier to see PR than discussing back
and forth to understand each others

Nebukadneza, 2018-06-07:
Sorry for the delay!

PR #619 (2018-06-15): Handle LocalizedText-Values in code generation
> [RFC] This attempt to fix #617 . I didn’t re-generate the address-space code yet, but simply wanted to show the change of the code-generator itself ….

Address-space code-generation didn’t handle nodes with LocalizedText in
Value-Attributes correctly before: It generated non-serializible
Attributes.

This fix redeems this so that proper LocalizedText objects are used,
which can be cleanly serialized.
closed 2018-06-15 by oroulet