FreeOpcUa/python-opcua #621: CryptographyDeprecationWarning

kleskjr, 2018-06-07:
uacrypto.py needs an update 
```
../opcua/crypto/uacrypto.py:58: CryptographyDeprecationWarning: signer and verifier have been deprecated. Please use sign and verify instead.
  hashes.SHA1())                                                                              
../opcua/crypto/uacrypto.py:48: CryptographyDeprecationWarning: signer and verifier have been deprecated. Please use sign and verify instead.
  hashes.SHA1()   
```

hirsche, 2024-02-27:
I have the same issue. Is this a bug or wrong package configuration?

AndreasHeine, 2024-02-28:
The library is deprecated.
Please switch to : [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio) which also has a [sync-wrapper](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/sync.py), with very few changes in API**

closed 2024-02-28 by AndreasHeine