FreeOpcUa/python-opcua #622: client error

akgaurihar, 2018-06-08:
we when executing the server have certain errors but we feel the code is all right . can anyone take a look of what is wrong in the code attached below .
messages are showed as follows;
"""""
client connected
Traceback (most recent call last):
  File "C:\Users\student\eclipse-workspace\opcua client\client.py", line 13, in <module>
    Temperature = Temp.get_value()
  File "C:\IP\lib\site-packages\opcua\common\node.py", line 146, in get_value
    result = self.get_data_value()
  File "C:\IP\lib\site-packages\opcua\common\node.py", line 155, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "C:\IP\lib\site-packages\opcua\common\node.py", line 261, in get_attribute
    result = `self.server.read(params)`
  File "C:\IP\lib\site-packages\opcua\client\ua_client.py", line 304, in read
    data = self._uasocket.send_request(request)
AttributeError: 'NoneType' object has no attribute 'send_request'

                                                                                                                              """""""""

[client.txt](https://github.com/FreeOpcUa/python-opcua/files/2084604/client.txt)

zerox1212, 2018-06-08:
Does it work with UA Expert?

akgaurihar, 2018-06-08:
server is working on ua expert, but when I run the client code, it is giving me this error

akgaurihar, 2018-06-08:
Thanks mate ,it worked on removing the get_value!!

zerox1212, 2018-06-08:
Then you only have the node object. I don't know why `get_value()` gives you a crash, but you could try `get_data_value()` instead.

Be aware, `get_value()` and `get_data_value()` only work on UA nodes of type Variable or Property. Make sure you `Temp` node is really something that has a value.

akgaurihar, 2018-06-11:
get_data_value did not work too pls find the the screen shots asked for!

[image]

[image]

[image]

oroulet, 2018-06-11:
data = self._uasocket.send_request(request)
    AttributeError: 'NoneType' object has no attribute 'send_request'

This is a very very strange error message. Either you try to get the value after the client is down or before.... try to sleep a second before get value. Let me know what happens. In theory it should not change anything, but one should never get that kind of error either...

oroulet, 2018-06-11:
try also to run your example outside of your editor. Somethime these ~crappy~advanced editor make very strange things in the background

akgaurihar, 2018-06-11:
It was strange, but  when we rewrote the same code it worked fine!

closed 2018-06-11 by akgaurihar

Kog008, 2021-07-05:
as far as i am understanding python and its concepts: the __exit__ function is automatically called. So your connection is already gone, when you are trying to close it. U can use the with statement instead: i.e.

with Client(server_url) as opc_client:
        print('Successfully connected opc reader to ', __server_url)
        root = opc_client.get_root_node()
        print("Object node is: ", root)

With this code the __enter__ & __exit__ functions of the Client class are called by default. U can see this by debugging.

MariosGkMeng, 2022-02-16:
I am facing the same problem here. I would like to add that this particular error happens right after the _uasocket attribute is called:

TIME.server._uasocket.

This expression leads to a "NoneType". 
I reckon that this has to do which a socket-related problem?