FreeOpcUa/python-opcua #623: Error starting Server

MikelAyani, 2018-06-13:
Hi,
I just updated the library with pip and get the next error when starting the server:
`add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15964), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16134), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16135), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists`
So far I think the problem is in standard_address_space_part5.
Regards,
Mikel

oroulet, 2018-06-13:
This is not important. Already fixed in master. Will be released soon

MikelAyani, 2018-06-13:
Ok, version 0.95.1 works so I downgraded.

oroulet, 2018-06-13:
You can test master by writing pip install opcua --pre --upgrade it should be much faster

oroulet, 2018-06-13:
And report bugs

MikelAyani, 2018-06-13:
ok, thanks!

Jayant-Pancholi, 2020-06-07:
Hi , I am facing this issue -
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
It says Parent node does not exists.
asyncua.ua.uaerrors._base.UaError: Parent of event type could not be found as well.
Please let me know about the possible fix for this issue.

kamarya, 2020-07-14:
same issue with `example_server` on 

```
commit d24d7806a71739b869f6d2465316184b49e37875 (HEAD -> master, origin/master, origin/HEAD)
Merge: f7c41d2 0310faf
Author: Denis Štogl <destogl@users.noreply.github.com>
Date:   Wed Jun 17 10:52:00 2020 +0200

    Merge pull request #361 from morxa/boost-173-std-algorithm-includes
    
    Add missing includes to algorithm header

```

here are the errors

```
[2020-07-14 10:07:14.987] [server] [error] address_space_internal| parent node 'ns=0;i=11489;' does not exists
[2020-07-14 10:07:14.987] [server] [error] address_space_internal| parent node 'ns=0;i=11489;' does not exists
[2020-07-14 10:07:15.009] [server] [error] address_space_internal| parent node 'ns=0;i=11580;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11580;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11583;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11585;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11585;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11588;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11590;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11590;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11593;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11629;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11629;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11632;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11634;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11634;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11637;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11639;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11639;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11642;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11680;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11680;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11683;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11685;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11685;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11688;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11690;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11690;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11693;' does not exists
[2020-07-14 10:07:15.048] [server] [error] address_space_internal| parent node 'ns=0;i=11492;' does not exists
[2020-07-14 10:07:15.048] [server] [error] address_space_internal| parent node 'ns=0;i=11492;' does not exists
[2020-07-14 10:07:15.077] [server] [error] address_space_internal| parent node 'ns=0;i=9029;' does not exists
[2020-07-14 10:07:15.077] [server] [error] address_space_internal| parent node 'ns=0;i=3875;' does not exists
[2020-07-14 10:07:15.079] [server] [error] address_space_internal| parent node 'ns=0;i=9069;' does not exists
[2020-07-14 10:07:15.080] [server] [error] address_space_internal| parent node 'ns=0;i=9111;' does not exists
[2020-07-14 10:07:15.080] [server] [error] address_space_internal| parent node 'ns=0;i=9113;' does not exists
[2020-07-14 10:07:15.082] [server] [error] address_space_internal| parent node 'ns=0;i=9213;' does not exists
[2020-07-14 10:07:15.085] [server] [error] address_space_internal| parent node 'ns=0;i=2949;' does not exists
[2020-07-14 10:07:15.128] [server] [info] Root node is: Node(ns=0;i=84;)
[2020-07-14 10:07:15.128] [server] [info] Children are:
[2020-07-14 10:07:15.128] [server] [info]     Node(ns=0;i=85;)
[2020-07-14 10:07:15.128] [server] [info]     Node(ns=0;i=86;)
[2020-07-14 10:07:15.128] [server] [info]     Node(ns=0;i=87;)
```

AndreasHeine, 2020-07-14:
`ns=0;i=11489;`

where does the last ";" came from?

kamarya, 2020-07-17:
this is when I run the compiled `example_server`. it is not Python.

swamper123, 2020-07-17:
It seems to be an issue in their code, because the semicolon is to much in the node ID. It would be interpreted the wrong way.

kamarya, 2020-11-06:
Any updates on this ? Is this a serious problem ? I still get this with both C++ and Python implementations:

```
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3062), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3063), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=24), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=31), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3068), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=12170), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3067), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3069), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3070), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11433), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11498), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15002), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=12908), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11512), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11513), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11432), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3071), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=12745), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3072), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=3073), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=16306), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=17605), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=23501), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=58), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=62), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=78), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=80), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=83), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11508), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11510), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=84), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
[... cut ...]
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11427), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11428), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(i=11429), requested parent node is null TwoByteNodeId(i=0) True
INFO:opcua.server.internal_server:Created internal session Internal
INFO:opcua.server.internal_server:starting internal server
WARNING:opcua.server.binary_server_asyncio:Listening on 0.0.0.0:4840
```

drypatrick, 2023-02-20:
> this is when I run the compiled `example_server`. it is not Python.

I get the same issue running the same file