FreeOpcUa/python-opcua #624: Server

shawnnasetechnotects, 2018-06-13:
Hi, 
When I try to start the server-minimal provided, I get the following issues:

add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=15964), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16134), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16135), requested parent node NumericNodeId(i=15957) does not exists
add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists
Listening on 10.0.0.46:4840

If you direct me to the source of this problem and give me some guidance in resolving it, I would appreciate it.
Thanks.

zerox1212, 2018-06-13:
Try using the code on the master branch. I think this is already fixed but isn't deployed to `pip`.

nem8, 2018-06-20:
Can confirm that installing from master branch fixes this issue.

Jayant-Pancholi, 2020-06-07:
Hi , I am facing this issue -
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
It says Parent node does not exists.
asyncua.ua.uaerrors._base.UaError: Parent of event type could not be found as well.
Please let me know about the possible fix for this issue.