FreeOpcUa/python-opcua #625: FileType Objects

MikelAyani, 2018-06-13:
Hi,
I can see that FileType object type is defined in the server, but I wonder if it is fully implemented and if someone has an example of how to use it, write/read operations...
Regards,
Mikel

zerox1212, 2018-06-13:
I do not think it's implemented unless the file is stored in memory as binary data. I have not read the OPC UA spec on how this should work either.

Note that nearly all types will be in the server because it's built from the OPC UA spec XML. That doesn't mean there is code to handle them all.

oroulet, 2018-06-14:
What do you mean by fully implemented? As far as I know a file type is just a data type and you encode it as bytestring I guess

zerox1212, 2018-06-14:
I don't remember the spec, was there file service for disk operations?

MikelAyani, 2018-06-14:
Thanks for the answers. What I ask is if the methods are implemented. I can see the objecttype is defined and its properties and methods included. But when I create an instance and call for example the Open method I get an error.

oroulet, 2018-06-14:
Object type is just a type you need to instantiate it somewhere in your
address space and implement the method

zerox1212, 2018-06-14:
I see. In that case you should be able to look at how methods are done in the example code and implement the `Open` method.

scareynz, 2018-09-07:
sorry so just to confirm, we have to define all of the methods for file type, such as read, write, open etc. in the python server?

Cheers

MikelAyani, 2018-10-02:
As I understand yes.

scareynz, 2018-10-03:
Thanks for all the help, this SDK worked brilliantly for our university project and wouldn't have been able to do it without the opensource community so thank you.

LilacBlue, 2018-10-19:
After reading your discussion I'm still very confused. I have little experience in Python and OPC UA in general, but I have to implement an OPC UA server which will offer the possibility to download pdf and graphical files. I couldn't find any example of such implementation. 
Is there any method such as:
`my_object.add_variable(idx, "My document", "myDocument.pdf", ua.VariantType.FileType)` 
that I could use for offering files or am I getting it all wrong and there is another way to do it? 
I will highly appreciate any help.

zerox1212, 2018-10-19:
You would have to read the OPC UA spec on how to handle files. I tried to go search the spec but OPC Foundation website is going really slow and I can't login.

As a work around you could use the existing library to make a `method` which returns a bytestring of the PDF. then your client can turn the bytestring back into a PDF file.

oroulet, 2018-10-19:
File type is not a binary type. Send your document as bytestring

MikelAyani, 2018-10-19:
Hi LilacBlue,
Something like this bellow does the trick for us.  We do not use the standard FileType, we create our own objecttype in which we add the next two methods (not exactly because I removed some specific code) and some properties. In the example bellow the FILE_PATH is hard-coded but you can use one of the properties to set the file path.
I hope it helps!

`    
FILE_PATH = 'C:/Temp/test.pdf'
    
    @uamethod
    def file_Get(self, nodeid:ua.VariantType.NodeId) -> bytes:
        """ Method to get a file.
        :param nodeid: file object nodeid
        :returns: File stream
        """
        try:
            with open(FILE_PATH, 'rb') as f:
                content = f.read()
            return content    
        except Exception as e:
            print('Failed to read file: {0}'.format(e))
 

    @uamethod
    def file_Write(self, nodeid: ua.VariantType.NodeId, file_stream:bytes) -> bool:
        """ Method to write a file.
        :param nodeid: file object nodeid
        :param file_stream: File content bytes stream
        :returns: Success
        """
        try:
            with open(FILE_PATH, 'wb') as f:
                f.write(file_stream)
            return True
        except Exception as e:
            print('Failed to write asset file: {0}'.format(e))
        return False
`