FreeOpcUa/python-opcua #626: creating subtypes and defining the definitions for properties

akgaurihar, 2018-06-14:
As part of my integrated project i need to take the data from ecl@ass website and define the subtyptes as shown in the pictures below.
1) how to code different subtypes or properties under 1 folder?
2) how to input definitions to certain properties /attributes?

[image]
[image]

I tried writing this way but iam not able to give multiple subtypes under one cetegorie.please check the screen shots in relation with code.

myobj=objects.add_object(idx, "AUTOMATION")
myobj=myobj.add_object_type(idx, "robotics assembly")
myobj=myobj.add_folder(idx, "robotics kinematics parts")
myobj=myobj.add_object_type(idx,"0173-1#02-AAO742#002 - Brand")
myobj=myobj.add_property(idx,"02-AAO742 Brand", 0)
myobj=myobj.add_variable_type(idx, "Part of the naming for the support and the recognition of the brand position of products and services consisting of words, numbers, letters or other characters. Registered brands and trademarks are indicated with the appropriate protective signs",STRING)

zerox1212, 2018-06-14:
I don't understand what you are trying to do, but normally you define custom types in the server first.
`dev = server.nodes.base_object_type.add_object_type(0, "robotics assembly type")`

Then when you instantiate a new object you set it to the type you defined.

akgaurihar, 2018-06-14:
our task is to define any example( in this case we chose kinematics parts )with all its properties  from the ecl@ss website as mentioned on 1 server and few other properties of the same example on another server and connect the 2 servers over a client.

we wanted to code in such a way that it just replicates the website for client to view it on ua expert.

link for refrence : http://www.eclasscontent.com/index.php?id=27389101&version=10_1&language=en&action=det

when u click " 0173-1#02-AAO742#002 - Brand " under kinematics parts it shows few properties like
Classification,name,properties,definition etc so we wanted to know how we can code this in such a way that we have these 4 properties under  " 0173-1#02-AAO742#002 - Brand ". check the ua expert screen shot attached below
[image]

zerox1212, 2018-06-14:
You are saying that you can't have multiple children under "0173-1#02-AAO742#002 - Brand"?

akgaurihar, 2018-06-14:
yes! we are not able to create subtypes like property, short name, format, defination under the title  "0173-1#02-AAO742#002 - Brand"
[image]

oroulet, 2018-06-14:
What happens? Do you get any error message?

akgaurihar, 2018-06-14:
no! we are not getting any errors, but we are not able to create subtypes

zerox1212, 2018-06-14:
That isn't how `types` work in OPC UA. To do what you want with types you would have to make many smaller types.

```python
# create our custom object type for brand
    brand = server.nodes.base_object_type.add_object_type(0, "Brand")
    brand.add_variable(0, "sensor1", 1.0)
    brand.add_property(0, "device_id", "0340")

# create our custom object type for robot
    assem = server.nodes.base_object_type.add_object_type(0, "Assembly")
    assem_folder = assem.add_folder(idx, "robotics kinematics parts")
    assem_folder.add_object(idx, "brand", brand_type)  #fix this, i don't remember how to add an object with data type at the moment

# create our custom object type
    assem = server.nodes.base_object_type.add_object_type(0, "Assembly")
    assem_folder = assem.add_folder(idx, "robotics kinematics parts")

# instantiate our new object type
    myrobot= instantiate(server.nodes.objects, assem, bname="2:assem0001")
```

oroulet, 2018-06-14:
@akgaurihar what does that mean? Either you get an error out you have some code not behaving like expected. To create a subtype you create a node type under an existing type. And that works fine

akgaurihar, 2018-06-15:
Thank you so much @zerox1212  and @oroulet  for your help, actually I am new to OPC UA and UA expert. I am learning and doing a project on the topic.

akgaurihar, 2018-06-15:
@zerox1212 now according to you , so in order to make sub types we need to code separately, as u have shown but when in u r code it doesnt show the subtypes in address space of ua expert ,it shows in references.

@oroulet yes the code is not behaving in the way we are expecting as told previously about the task our addresss space should display all the sub types according to the website of ecl@ss. please take a look at my last screenshot , so i click  "0173-1#02-AAO742#002 - Brand" i need to have 5 sub types(property,format,shortname etc) shown on my address space barof ua expert moveover which  clicked need to display the data shown in the screen shot on attributes or references section.
 I just wanted to know if this is possible or not.

zerox1212, 2018-06-15:
The`type` nodes are listed under the server object. Look under server>data types>base object type or whatever.

That is what I was saying earlier, your application isn't a good fit for types because your structure doesn't have anything that is reusable. You might as well just build your entire tree with normal objects.

An example of data type would be like this:
```
DataType for Machine:
  Machine
    >ID
    >Temperature
    >Status
```
Then if you had several machines you could instantiate this structure and get this:
```
myMachine1
  >ID = 101
  >Temperature = 36
  >Status = Running

myMachine2
  >ID = 105
  >Temperature = 24
  >Status = Stopped
```

yangpeiren, 2018-06-17:
@zerox1212 , could you please give an example of the customized DataType in code? I've searched in the project, only found examples in an xml file named enum_struct_test_nodes.xml under test folder, the customized DataType has such part
        <Definition Name="1:ExampleStruct">
            <Field DataType="Int16" Name="IntVal1"/>
            <Field DataType="ExampleEnum" Name="EnumVal"/>
        </Definition>
could you please give an example of how to create this in code? Thank you very much!

zerox1212, 2018-06-18:
An `enum` and a `type` are not the same in OPC UA.

There are already examples... 
Create a type: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-custom-object.py
Create a enum: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-enum.py

oroulet, 2018-06-18:
@zerox1212 good example ;-) . On client side (or if enums are defined in XML)) one can use  load_enums() now that we have that function instead of creating Python enum en by hand.

zerox1212, 2018-06-18:
Do we need to update the `enum` example?

oroulet, 2018-06-18:
No