FreeOpcUa/python-opcua #627: utils.ThreadLoop in InternalServer not re-initialized in start⇒stop⇒start scenario

Nebukadneza, 2018-06-15:
`InternalServer` creates a `utils.ThreadLoop()` for `asyncio` handling. This `Thread` is started and runs until the server stops. However, if you re-start the server (which is possible / not catched), the `ThreadLoop` object is not re-intialized. This leads to a `RuntimeError: threads can only be started once`, because `ThreadLoop` is-a `Thread`, and threads that exited once cannot be re-started.

To fix this, a new `ThreadLoop` needs to be created when re-starting the server. This new object also needs to be handed to `BinaryServer` again to update its reference, which is still on the old object.

Finally, the `loop` is currently handed to the `SubscriptionService`, which never uses it — so we can get rid of the additional bogus reference to the object.

I’m opening a PR to fix this.

Thanks — and please tell me if my chain of thought is right ^_^.

oroulet, 2018-06-15:
Ok. Send or for restart. But I am quite sure the loop is used in InternalSubscription

Nebukadneza, 2018-06-15:
Hej,

sorry if i mis-read the code — i couldn’t find a reference to being used in SubscriptionService, but if i was worng, i’ll gladly change it so i can also update the loop-reference in SubscriptionService ….

Lets see what travis says ^_^.

Nebukadneza, 2018-06-15:
… you are right and i am blind. Sorry! I’ll fix this ^_^.

Nebukadneza, 2018-06-28:
Hi again,

sorry for the fuckup / wasting your time with the last attempt. This looks better IMO. Please let me know what you think ^_^.

Best
-Dario

PR #638 (2018-07-17): Fix re-initialization of ThreadLoop in InternalServer
#### Fixes #627 
Sorry this is a new PR — for some reason GitHub greyed out the »reopen« button on the old PR.

The various Server objects use a utils.ThreadLoop to handle the
asyncio-mainloop. This Thread(~like) object is not getting
re-initialized in a start()⇒stop()⇒start()-scenario, which leads to a
»Threads may only be started once« exception.

To fix this, we dump the old ThreadLoop object upon re-initialization,
and hand the new object around.
closed 2018-07-17 by oroulet