FreeOpcUa/python-opcua #629: AttributeError: 'NoneType' object has no attribute 'send_request'

Summer88, 2018-06-15:
Just running these row on a Debian 9

Python 3.5.4 (default, Aug 12 2017, 14:08:14) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

```
In [1]: from opcua import Client
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled

In [2]:  client = Client("opc.tcp://ciccio:62841/gigio/uaPLUS")

In [3]: var = client.get_node("ns=2;s=gigi.gigio.giuggiolo.PROGRAM")

In [4]: var.get_value()```

```
AttributeError                            Traceback (most recent call last)
<ipython-input-4-1144b4710b9d> in <module>()
----> 1 var.get_value()

~/.virtualenvs/x/lib/python3.5/site-packages/opcua/common/node.py in get_value(self)
    144         An exception will be generated for other node types.
    145         """
--> 146         result = self.get_data_value()
    147         return result.Value.Value
    148 

~/.virtualenvs/x/lib/python3.5/site-packages/opcua/common/node.py in get_data_value(self)
    153         DataValue contain a variable value as a variant as well as server and source timestamps
    154         """
--> 155         return self.get_attribute(ua.AttributeIds.Value)
    156 
    157     def set_array_dimensions(self, value):

~/.virtualenvs/x/lib/python3.5/site-packages/opcua/common/node.py in get_attribute(self, attr)
    259         params = ua.ReadParameters()
    260         params.NodesToRead.append(rv)
--> 261         result = self.server.read(params)
    262         result[0].StatusCode.check()
    263         return result[0]

~/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/ua_client.py in read(self, parameters)
    302         request = ua.ReadRequest()
    303         request.Parameters = parameters
--> 304         data = self._uasocket.send_request(request)
    305         response = struct_from_binary(ua.ReadResponse, data)
    306         self.logger.debug(response)

AttributeError: 'NoneType' object has no attribute 'send_request'
```
From opcua-client and uaexpert everything seems to work.

What's missing?
Thanks, BR.

Summer88, 2018-06-15:
missing a crucial step:
`
 client.connect()
`

closed 2018-06-15 by Summer88

amisha218, 2019-06-03:
from opcua import Client
from random import randint
import time
from time import sleep
import socket
url="opc.tcp://10.17.13.40:48402"
host="opc.tcp://10.17.13.40"
port=48402
client = Client(('10.17.13.40'),port)
def connect(self):
    client.connect(self)
print("Client Connected")
while True:
    temp=client.get_node("ns=2;i=3")
    Temprature= temp.get_value()
    print(Temprature)

    press=client.get_node("ns=3;i=4")
    pressure=press.get_value()
    print(press)

    time=client.get_node("ns=4;i=5")
    time=time.get_value()
    print(time)

    sleep(randint(2,2))
client.close()