FreeOpcUa/python-opcua #631: Adding reference fails with ua.ObjectIds but works with server.get_node(ua.ObjectIds).nodeid

NitashEU, 2018-06-19:
Hi Friends,

I am trying to add and delete references to a created variable.

```python
var = object_type.add_variable(namespace_id, "BrowseName", 0.0, None, ua.ObjectIds.Float)
```

When I try to use 

```python
var.add_reference(ua.ObjectIds.AnalogItemType, ua.ObjectIds.HasTypeDefinition)
```

it fails with following error:

```error
Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/usr/local/lib/python3.6/site-packages/opcua/server/uaprocessor.py", line 90, in process
    return self.process_message(msg.SecurityHeader(), msg.SequenceHeader(), msg.body())
  File "/usr/local/lib/python3.6/site-packages/opcua/server/uaprocessor.py", line 110, in process_message
    return self._process_message(typeid, requesthdr, algohdr, seqhdr, body)
  File "/usr/local/lib/python3.6/site-packages/opcua/server/uaprocessor.py", line 210, in _process_message
    self.send_response(requesthdr.RequestHandle, algohdr, seqhdr, response)
  File "/usr/local/lib/python3.6/site-packages/opcua/server/uaprocessor.py", line 44, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId, algohdr=algohdr)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 253, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 293, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 293, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 281, in to_binary
    return struct_to_binary(val)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 253, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 293, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 293, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 281, in to_binary
    return struct_to_binary(val)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 258, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 271, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 187, in pack_uatype
    return nodeid_to_binary(value)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 301, in nodeid_to_binary
    data = struct.pack("<BB", nodeid.NodeIdType.value, nodeid.Identifier)
struct.error: ubyte format requires 0 <= number <= 255
```

Getting the nodeids directly however works:

```python
has_type_definition_node = server.get_node(ua.ObjectIds.HasTypeDefinition)
analog_item_type_node = server.get_node(ua.ObjectIds.AnalogItemType)
var.add_reference(analog_item_type_node.nodeid, has_type_definition_node.nodeid)
```

NitashEU, 2018-06-19:
Thank you for your quick response.

Unfortunately `ua.NodeId(ua.ObjectIds.Float)` doesn't work either. I guess I'll have to stick with my "workaround".

Also is there any possibility to change the TypeDefinition of a variable, property without having to remove and add a reference? Also to apply for the parent as well?

zerox1212, 2018-06-19:
You should set a break point in your code and inspect what is the difference between the objects when you have `my_node.nodeid` vs. `ua.NodeId(ua.ObjectIds.Float)`. Maybe it needs to be `TwoByteNodeId()` or something.

I don't really understand your second question. I think what you want is to define a custom type in the server, then you can use `instantiate` to create an instance of your custom type. This instance will already have the proper references.

NitashEU, 2018-06-20:
Hi, 

sorry for the confusion. The second question was refering to the typedefinition of a variable. If I add a variable to an object or an object type, the type definition is per default `BaseDataVariableType`. The variable itself has a reference `HasTypeDefinition` where it says `BaseDataVariableType` and the object has a reference `HasComponent` pointing to the variable and saying `BaseDataVariableType`. 

When creating a variable I cannot define the typedefinition by myself. Thats why I manually remove the references and add the new typedefinition references.

Regards

zerox1212, 2018-06-20:
You have to ask @oroulet for that. I never had a reason to work with anything but the base.

oroulet, 2018-06-20:
yes it looks like this is harcoded in the node interface. I am not very keen into adding one more argument to these methods. delete and more typedefiniton of you only need to do this a few times. Otherwise look at code in _create_variable in opcua/common/manage_nodes.py and write your own method/function