FreeOpcUa/python-opcua #632: IdentityCriteriaType

PaBraIgm, 2018-06-19:
This is used in IdentityMappingRuleType and initialized to 0 in __init__, which does not exist in the enum.

oroulet, 2018-06-20:
is this a problem? this is just a type

PaBraIgm, 2018-06-26:
It raises an exception, so it seems to be a problem.

oroulet, 2018-06-26:
When/where do you get an exception? Show code and stack tracr

PaBraIgm, 2018-06-26:
Test Code

```
from opcua import ua
mapping = ua.IdentityMappingRuleType()

```
Stacktrace:

```
Traceback (most recent call last):
  File "d:\Projekte\OPC-UA\Test\FreeOpcUa\err.py", line 2, in <module>
    mapping = ua.IdentityMappingRuleType()
  File "C:\Pyhton36\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 1209, in __init__
    self.CriteriaType = IdentityCriteriaType(0)
  File "C:\Pyhton36\lib\enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "C:\Pyhton36\lib\enum.py", line 533, in __new__
    return cls._missing_(value)
  File "C:\Pyhton36\lib\enum.py", line 546, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 0 is not a valid IdentityCriteriaType

```

It is pretty easy to fix manually (just add something with value 0 to the enum). But it is part of autogenerated code, so that might not be the way to go....