FreeOpcUa/python-opcua #633: connect to server with security result in BadCertificateUriInvalid

Summer88, 2018-06-20:
I'm trying to connect with security to a server, but getting this error.
The same connection without security works fine.

Maybe I'm wrong but I believe the problem is that certificate got the name of the server not it's ip.
But from the UA expert the same settings works fine.
I'm over the network so I need to call the server by its IP address not by name.

```
In [7]: client = Client("opc.tcp://directipaddress:62841/ciccio/bello")
   ...: client.set_security_string("Basic256,Sign,/path/uaexpert.der,/path/uaexpert_key.pem")

In [8]: client.connect()
/home/tec1/.virtualenvs/backled3/lib/python3.5/site-packages/opcua/crypto/uacrypto.py:48: CryptographyDeprecationWarning: signer and verifier have been deprecated. Please use sign and verify instead.
  hashes.SHA1()
/home/tec1/.virtualenvs/backled3/lib/python3.5/site-packages/opcua/crypto/uacrypto.py:58: CryptographyDeprecationWarning: signer and verifier have been deprecated. Please use sign and verify instead.
  hashes.SHA1())
---------------------------------------------------------------------------
BadCertificateUriInvalid                  Traceback (most recent call last)
<ipython-input-8-be08c18772f3> in <module>()
----> 1 client.connect()

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/client.py in connect(self)
    253             self.send_hello()
    254             self.open_secure_channel()
--> 255             self.create_session()
    256         except Exception:
    257             self.disconnect_socket()  # clean up open socket

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/client.py in create_session(self)
    371         params.RequestedSessionTimeout = 3600000
    372         params.MaxResponseMessageSize = 0  # means no max size
--> 373         response = self.uaclient.create_session(params)
    374         if self.security_policy.client_certificate is None:
    375             data = nonce

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/ua_client.py in create_session(self, parameters)
    246         request = ua.CreateSessionRequest()
    247         request.Parameters = parameters
--> 248         data = self._uasocket.send_request(request)
    249         response = struct_from_binary(ua.CreateSessionResponse, data)
    250         self.logger.debug(response)

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/ua_client.py in send_request(self, request, callback, timeout, message_type)
     76         if not callback:
     77             data = future.result(self.timeout)
---> 78             self.check_answer(data, " in response to " + request.__class__.__name__)
     79             return data
     80 

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/client/ua_client.py in check_answer(self, data, context)
     85             self.logger.warning("ServiceFault from server received %s", context)
     86             hdr = struct_from_binary(ua.ResponseHeader, data)
---> 87             hdr.ServiceResult.check()
     88             return False
     89         return True

/home/tec1/.virtualenvs/x/lib/python3.5/site-packages/opcua/ua/uatypes.py in check(self)
    233         """
    234         if not self.is_good():
--> 235             raise UaStatusCodeError(self.value)
    236 
    237     def is_good(self):

BadCertificateUriInvalid: The URI specified in the ApplicationDescription does not match the URI in the certificate.(BadCertificateUriInvalid)
```

oroulet, 2018-06-20:
you can set the application uri in client object and you can do whatever you want with your certificate. That should solve it....in worst case use wiresharck to see all packets back and forth

mikelga, 2018-12-12:
Hello!
I think that I have a similar problem.
I want to communicate with the UaServerC++ (from Unified Automation) OPC-UA server with **Basic256Sha256** and **SignAndEncrypt**.
Without any security it let me connect properly:
[image]
[image]

But, then, when I add set_security_string it gives me a **The URI specified in the ApplicationDescription does not match the URI in the certificate.(BadCertificateUriInvalid)** error. I proved properly with same certificates files in UaExpert client, without any errors.
[image]
[image]

It's the problem that I have to set the application uri in client object? If the response is yes, how can I do this?

Thank you very much,

mikelga, 2018-12-12:
@oroulet Sorry for not mentioning you in the previous message

moimart1, 2019-01-14:
Take a look here https://github.com/FreeOpcUa/python-opcua/issues/778#issuecomment-454133568
Specially the "Note" part

nobodyman1, 2019-03-01:
I did the same thing: I used the certificates from UA Expert Client.
The solution is quite simple:

- Use this shell command to check the ApplicationUri:<br>``openssl x509 -in uaexpert.der -inform der -text -noout | grep URI``<br>Look at the string which starts with ``urn:``
- Add this line to your python script: ``print(client.application_uri)``<br>The output is ``urn:freeopcua:client``.
- Compare the two strings. If they are not equal, you have to add<br>``client.application_uri = "urn:<output-from-certificate-check>"``<br>to your python script. Insert it **before** you call ``client.connect()``.

By the way there is no need to use ``client.load_client_certificate()`` and ``client.load_private_key()`` if you use function ``client.set_security_string()``.

mikelga, 2019-03-01:
Hi @nobodyman1 !
Do you know any way to authenticate de user using certificates? Some certificates different to those used in the security settings. I want to use certificates to authenticate (instead of anonimously or username/password).
Thanks!

nobodyman1, 2019-03-01:
Hi @mikelga!
I don´t know because I´m connecting to a Siemens PLC S7-1500 which (as far as I know) doesn´t support certificates for user authentication only username and password or anonymous.
But if I take a look at the documentation [https://python-opcua.readthedocs.io/en/latest/client.html](url) I think the functions you are using (``load_client_certificate(path)`` and ``load_private_key(path)``) are correct.

- Did you try to enable debug output to get some hints?
- Did you try another OPC UA Client before to connect to your server?
- Did you try to connect using username and password?