FreeOpcUa/python-opcua #634: Concurrent Access of UA Server

zhaoyf614, 2018-06-20:
I wrote an OPC UA Server similar to the demo, but I found that this server does not supoort methods executed in paralell, even if i call them in paralell, the ua_methods will end up executing in paralell, what should I do? Is there any settings or function in the server side for this?

oroulet, 2018-06-20:
well you have a few fundamental issues here:
* the server is based on asyncio so everything is executed sequencially
* It might in theory be possible to use threads (an earlier version of our server was based on thread), but anyway this is a python server. a python application is always a one process application so you will never get real parallelism with python

it might in theory be possible to use multiprocess but it would be hard work to find out how to make this efficient...

zerox1212, 2018-06-20:
I would be surprised if any OPC UA servers support method parallelism.