FreeOpcUa/python-opcua #635: send a file to server and back to client

RasPiTool, 2018-06-21:
Hello, 
Thanks for python-opcua!
I have a fmu-file or zip-file which contians the compiled simulation file.
Now I would like to send this fmu-file or zip-file from PC (Client) to Raspberry PI (Server). 
The Server should unzip the file and execute the simulation.
Unzipping and simulation on Raspberry Pi works fine.
But how can i send the file to the Server and after simulation back to the client?

oroulet, 2018-06-21:
Either make a ua method or 2 variables of type bytestring. Subscribe to first one on server side, wrote from client, and when new value arrive, simulate then write result in the other variable, etc

closed 2018-09-03 by RasPiTool

saimanoj567, 2020-11-22:
Can anyone post sample code to access the files from server via opc ua ?

AndreasHeine, 2020-11-22:
issue is closed!