FreeOpcUa/python-opcua #636: How do I connect to a kepware server?

rockyfish44, 2018-06-21:
I'm having a few issues here.... Sorry in advance, definitely a novice; but I'm attempting to get my toes in the water hear.

I have the port connection from KepServer, where do I put this? Do I have to generate a certificate for the OPC Python client? If so, how? I was unable to figure out how to use the generate_certificate.sh file.

Finally, I was able to start up the demo OPC UA server in the command line, as well as connect the client to it (it automatically does I think?). How do I the navigate through the server nodes in the client?

Thanks for your time.

zerox1212, 2018-06-21:
Are you using command line tools or writing a client?

rockyfish44, 2018-06-21:
Ideally I would like to write one.... but starting out with the command line tool seems simplest. In the end, I would like to extract data from a server with a client, and output it to the screen continually.

Which documentation should I read? I read the README of course, but for further info, where should I look?

zerox1212, 2018-06-21:
If you want to do that I would download pyCharm and start trying to get a simple client example to read a node from your kep server.
The easiest way to get a visual idea of what you need to do is to inspect the server data structure with UA Expert. Then you can use node methods to get to the same data with a python client.

http://python-opcua.readthedocs.io/en/latest/