FreeOpcUa/python-opcua #637: Variable with differeing VariantType and DataType

helmut-jacob, 2018-06-22:
I can easily create variable nodes where the datatype attribute differs from the actual variant datatype of the datavalue. If I read the spec correctly this shouldn't be possible.

From  OPC Unified Architecture, Part 3, 5.6.2:
> The most recent value of the Variable that the Server has. Its data
> type is defined by the DataType Attribute. It is the only Attribute that
> does not have a data type associated with it. This allows all Variables
> to have a value defined by the same Value Attribute.

Here's a simple code snippet that generates an invalid variable node.
```
import opcua
s=opcua.Server()
x=s.get_root_node().add_variable(nodeid=100,  bname="var", val=0, datatype=opcua.ua.ObjectIds.Double)

In [9]: x.get_data_type_as_variant_type()
Out[9]: <VariantType.Double: 11>

In [11]: x.get_data_value()
Out[11]: DataValue(Value:Variant(val:0,type:VariantType.Int64), StatusCode:StatusCode(Good), SourceTimestamp:2018-06-22 13:31:42.999098, ServerTimestamp:2018-06-22 13:31:42.999097)
```

Additionally set_value will guess the variant datatype from the value and won't leverage datatype information.

Not sure what the right approach would be but I would favor to force the variant datatype to be the same as the datatype attribute and bail out with an exception if a conflict is detected (for example in set_value).

Any opinions?

zerox1212, 2018-06-22:
I think this has been brought up before. I don't remember the outcome, however I think the spec should not be fully enforced on the server side here.

What you are doing on the server side is up to you, but if you make a strongly typed variable it will be enforced for clients.
`myarrayvar = myobj.add_variable(idx, "myStronglytTypedVariable", ua.Variant([], ua.VariantType.UInt32))`

helmut-jacob, 2018-06-22:
I could of course ensure those semantics outside of the library, still I think we should try to prevent invalid behavior. Something like the following maybe?

```
diff --git a/opcua/common/node.py b/opcua/common/node.py
index d7db9b5..b3debd6 100644
--- a/opcua/common/node.py
+++ b/opcua/common/node.py
@@ -198,8 +198,11 @@ class Node(object):
             datavalue = value
         elif isinstance(value, ua.Variant):
             datavalue = ua.DataValue(value)
-        else:
+        elif varianttype:
             datavalue = ua.DataValue(ua.Variant(value, varianttype))
+        else:
+            datavalue = ua.DataValue(ua.Variant(value, self.get_data_type_as_variant_type()))
+
         self.set_attribute(ua.AttributeIds.Value, datavalue)
 
     set_data_value = set_value

```

oroulet, 2018-06-22:
This is not so simple. There is not one to one correspondence between variant type and datatype. You can define whatever new data type with variant type Boolean for example. When creating a node variable we try to give sensible default value but it cannot be right all the time