FreeOpcUa/python-opcua #639: Static variable typing

AlexEshoo, 2018-06-29:
Is there a way currently to specify the data type of a node in a static way? My server updates values from python objects and, for instance, when it writes an int, the datatype shown in UaExpert is `Int64`. Is there a way I can set this to `Int16` and have it stay that way?

I looked through the node module but it seems pretty confusing to me the way it is structured. If I'm correct, there are only 6 different `NodeIdType`s the one being used by default is the `NodeIdType.Numeric`. I see that there are ones called `TwoByte` and `FourByte` do these represent integers?

Also, I have some float values that I will need to put in the server as well, is there support for a float type?

oroulet, 2018-06-29:
Look in example folder. Many examples for that mynode.set_value(3, ua.varianttype.UInt32) for example ( if I remember well)

AlexEshoo, 2018-06-29:
Thanks for your quick response. It does seem to change the datatype shown in UaExpert, at least in one location, but I'm confused about what I'm seeing now:
[image]

The data type looks right in the "Data Access View", but the DataType Attribute is still set to Int64. Also if I try to read back the data type in the python program using `node.get_data_type_as_variant_type` it also returns `VariantType.Int64`.

Any idea what the discrepancy is?

oroulet, 2018-06-29:
When creating a variable you can also pass a type so you get correct data type

AlexEshoo, 2018-06-29:
Thanks again, that worked. Interesting that the data type of the node is allowed to be inconsistent with the data type of the value. Is that intended behavior? Is there a way to require them to be consistent?

oroulet, 2018-06-29:
This is a performance issue

AlexEshoo, 2018-06-29:
I'm not sure I understand what you mean, can you elaborate?

zerox1212, 2018-07-02:
He means that checking if the node data type matches the variant type every time a node is accessed will reduce performance. If you really need to enforce the data type you should handle it yourself.

AlexEshoo, 2018-07-02:
I see. Okay I will look into that, thanks again for your help.

zerox1212, 2018-07-02:
You should also understand that enforcing "strong" types will probably never work that well simply because Python is not designed to have strongly typed variables. This feature is part of what makes Python easy to work with.

AlexEshoo, 2018-07-02:
Yeah, I understand that python allows dynamic typing, but for something like OPC-UA it may be very important to some of the interfacing programs/devices what the data type is. For example if I write a driver which takes in the data type stored in the server and write it to modbus, it had better be a Uint16 or I'm gonna have problems. I would at least want the server to throw an exception or a warning if the incorrect data type is written to it.

oroulet, 2018-07-02:
As I wrote earlier this is a performance issue. Many applications do a lot of write at high frequency and checking for datatype is complex derfor expensive. Smart ideas are welcome to reduce cost of write and maybe add extra check but this will require a very good understanding of protocol and implementation

oroulet, 2018-07-02:
Caching the type in our main dict is a possibility but it has also a cost

AlexEshoo, 2018-07-02:
Perhaps theres a misunderstanding, I'm only concerned with either the server or client module of this package throwing an error if the datatype is being changed. If a low level device writes in a value that is incorrect I agree it will be expensive and probably impossible to tell. I am only asking that the API for the client and server be able to detect the discrepancy.

Consider the following:

On the server a variable is defined as such:
```python
myVar = someNode.add_variable(node_id, tag_name, 0, varianttype=VariantType.UInt16)
```
And then the client (implemented using this module) attempts something like this:

```python
myVar.set_value(val, VariantType.Int64)
```

I would want to prevent that from happening. Perhaps it is too expensive to cache it in the main dict object so I might just have my client application check what the `variantType` of the node is before writing and throw it's own exception if it doesn't agree.

zerox1212, 2018-07-02:
If the server node is defined with a strong type I think the client will get an error if it tries to write the wrong type. Does this not happen?

The issue you show I have only seen when the server writes to its own node (in which case the user should make sure their own server is not mixing types).

AlexEshoo, 2018-07-03:
I just tested with a separate server and client. The server defines the variable as:
```python
myvar = myobj.add_variable(idx, "MyVariable", 6.7, varianttype=VariantType.UInt32)
```

and the client sets it as:
```python
var = root.get_child(["0:Objects", "2:MyObject", "2:MyVariable"])
var.set_value(4, VariantType.UInt16)
```

There is no exception thrown, and the value data type is changed.

oroulet, 2018-07-03:
The code and API on client and server is the same. If we want to add type testing only for clients, this must probably be done in the network layer of server. Uaprotocol.py I suppose... Or user the internalSession to bypass it... But none of them has any solution for caching something so far

xaver-k, 2018-07-19:
Hi, I ran into the same problem and did a little digging around in the OCP-UA specifications. After what I can find, it looks like type-testing is not the task of the client. Instead, it seems to be a mandatory function for the server.

Specification Part 4 (Services) v1.04 states in Section 5.10.4.2 / Table 59:
```
A Server shall return a Bad_TypeMismatch error if the data type of the written
value is not the same type or subtype of the Attribute’s DataType. Based on the
DataType hierarchy, subtypes of the Attribute DataType shall be accepted by the
Server.
```

I also did some testing and could overwrite e.g. Double values with strings on a free opcua server. Testing with other server implementations like the one at `opc.tcp://opcua.demo-this.com:51210/UA/SampleServer`, actions like this get rejected by the server.

oroulet, 2018-07-19:
For reference if someone wants to try implementing this: there is already a method to find variant type given a node datatype: https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/ua_utils.py#L205

but that method uses the high level API. Unfortunately the correct place to implement the check is probably  here :  https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/address_space.py#L53  and so deep in server that API is not available

I am really not sure how and where to implement such a check....