FreeOpcUa/python-opcua #642: List servers hosted in one IP

Tenceto, 2018-07-02:
Hi all, sorry if this question is far too basic, I'm just getting started with OPC and, in particular, with this library.

I'm wondering if it is possible to list all servers that are being hosted in a given IP. As far as I understand, when running the following script:

`from opcua.tools import uadiscover
uadiscover()`

The function automatically looks for servers at `127.0.0.1:4840`.

Isn't there a way to do this for generic IPs and generic ports?

oroulet, 2018-07-02:
As far as I remember you can specify IP address

Tenceto, 2018-07-02:
How can I do that? When I execute the line `uadiscover()`, it automatically prints:

`Performing discovery at opc.tcp://localhost:4840`

And if I try to pass an IP address to the function, like `uadiscover("192.168.1.5")`, it returns:

`TypeError: uadiscover() takes no arguments (1 given)`

oroulet, 2018-07-02:
Uadiscover is not meant to be used directly. It is the code of a command line called uadiscover. Look at its code or use the command line

closed 2018-07-03 by Tenceto