FreeOpcUa/python-opcua #644: Server: No model change events are sent to client

simmarn, 2018-07-04:
My team have implemented a server where objects (folders) and variables (nodes) are added at run time. We are able to add nodes, but neither UAExpert nor our own UA based client gets any events when nodes are added. Client needs to re-browse to see the new nodes.

Is there any way to trigger model change events to clients when nodes are added? I have not found anything in documentation and examples for this.

oroulet, 2018-07-05:
The model changed event is not fired automatically but it should be easy to instanciate such an event and trigger it from server node. Look at the events examples and tell me how it goes. This is maybe something to document/make example for

simmarn, 2018-07-05:
I have scanned through the examples. Is it a custom event I need to generate? 
I am looking for an example were a folder or node is sent as an event. But I can not find anything.

Please point me in the right direction.

oroulet, 2018-07-05:
Look at server_example.py. call get_event_generator() but pass the node id
of ModelChangeEvent node (somewhere in address space under types). You will
get an object. You can then set an object member called ModifiedNodeId to
the nodeid of of folder that has changed or similar then trigger event. ( I
am guessing all this so names may be wrong but the process should be about
as I wrote)

simmarn, 2018-07-05:
Thanks for your help. I am getting closer.
I can not find member ModifiedNodeId. Is it add_properties that shall be used?

oroulet, 2018-07-05:
All properties of object are supposed to be set from the node type given as arg. I can have a look later today

oroulet, 2018-07-05:
I checked and this is a bit more complicated. The variable is called
'Changes' and should contain a list of ua.ModelChangeStructureDataType
which itself has a few members to fill up. Looks at table 157 in opcua spec
part 5... Use ua.ModelChangeStructureVerbMask to set correct bytes in the
'verb' member

Would be great if you could submit an example when you make this work

>
>

simmarn, 2018-07-06:
I tried as below. But I do not get any event in the client. Is it something I have missed when using the eventgenerator?

	entity = objects.add_object(idx, "Entity")
	entity.add_variable(idx, "SerialNumber", entity_data["SERIALNUMBER"])
	entity.add_variable(idx, "Family", entity_data["FAMILY"])
	# Create model change event
	evgen = server.get_event_generator(2133, entity)
	changes = []
	added_verb = ua.ModelChangeStructureVerbMask.NodeAdded
	entitychange = ua.ModelChangeStructureDataType()
	entitychange.Affected = entity.nodeid
	entitychange.AffectedType = entity.get_type_definition()
	entitychange.Verb = added_verb
	changes.append(entitychange)
	childnodes = entity.get_children()
	for node in childnodes:
		nodechange = ua.ModelChangeStructureDataType()
		nodechange.Affected = node.nodeid
		nodechange.AffectedType = node.get_type_definition()
		nodechange.Verb = added_verb
		changes.append(nodechange)
	evgen.Changes = changes
	evgen.trigger()

simmarn, 2018-07-06:
get_children() did not work as I though so I improved a bit. But I still do not get any event.

	entity = objects.add_object(idx, "Entity")
	childnodes = []
	childnodes.append(entity)
	childnodes.append(entity.add_variable(idx, "SerialNumber", entity_data["SERIALNUMBER"]))
	childnodes.append(entity.add_variable(idx, "Family", entity_data["FAMILY"]))
	# Create model change event
	evgen = server.get_event_generator(etype=2133, source=entity.nodeid)
	changes = []
	changes.append(entitychange)
	for node in childnodes:
		nodechange = ua.ModelChangeStructureDataType()
		nodechange.Affected = node.nodeid
		nodechange.AffectedType = node.get_type_definition()
		nodechange.Verb = added_verb
		changes.append(nodechange)
	evgen.Changes = changes
	evgen.trigger()

oroulet, 2018-07-06:
I haven't checked the spec but I am rather sure that the emitting node
should be the server node, so do not provide second argument to
get_event_generator()

simmarn, 2018-07-06:
Thank you. That gets me a bit further, but still no event. I get "Event %s does not fit WhereClause, not generating event". 

I guess the Select and WhereClauses on client side does not match the event, which result in the event not being generated. I guess I need to supply more information in the event, but I do not know how.

oroulet, 2018-07-06:
This is evgen.event. use introspection! for example in ipython

simmarn, 2018-07-06:
Thanks. Will do. Right now I am on vacation. I will continue when I get back to work in August.

oroulet, 2018-07-06:
I just did a small test. Looks like nobody ever tested a custom event containing members of custom datatype and this is obviously broken,,,

oroulet, 2018-07-07:
Great. Help (especially PR) is welcome. BTW if you consider going into production with ua I may be able to help I have an email address in setup.py as far as i remember

Lukas1811, 2019-09-02:
I have the same problem did you found a solution for that ?

zerox1212, 2019-09-02:
I don't think there was ever a PR for this. As a work around you could make a custom event that your client will subscribe to in which you could use the event to trigger the client to rebrowse. Otherwise you need to make a PR to custom datatypes inside of an event.

Lukas1811, 2019-09-02:
Ok, thank you, the client version wouldn't work out for me because it should support multiple different clients. I would try my best to find another solution. If I find something I will post it, that everyone could use it

simmarn, 2019-09-02:
Hi,

What we ended up doing was to send a custom event with all new node information sent in a json formatted string. Like this:

```
##returns a dictionary of names and ids
def create_node_data(node, parent):
	data = {}
	data["NodeName"] = node.get_browse_name().Name
	data["NodeId"] = node.nodeid.Identifier
	data["NodeNamespaceIndex"] = node.nodeid.NamespaceIndex
	data["SourceNodeName"] = parent.get_browse_name().Name
	data["SourceNodeId"] = parent.nodeid.Identifier
	data["SourceNodeNamespaceIndex"] = parent.nodeid.NamespaceIndex
	return data
	
##creates an entity in the object-node,
#containing all nodes specified in
#the specification, using the values in the entity_data dictionary
def create_entity(entity_data, objects, idx, server):	
	node_data = []
	
	entity = objects.add_object(idx, "Entity")
	node_data.append(create_node_data(entity, objects))
	serial_number = entity.add_variable(idx, "SerialNumber", entity_data["SERIALNUMBER"])
	node_data.append(create_node_data(serial_number, entity))
	family = entity.add_variable(idx, "Family", entity_data["FAMILY"])
	node_data.append(create_node_data(family, entity))
	type = entity.add_variable(idx, "Type", entity_data["TYPE"])
	node_data.append(create_node_data(type, entity))
	name = entity.add_variable(idx, "Name", entity_data["NAME"])
	node_data.append(create_node_data(name, entity))
	checked_in = entity.add_variable(idx, "CheckedIn", (entity_data["CHECKEDIN"] == "True"))
	node_data.append(create_node_data(checked_in, entity))
	status = entity.add_variable(idx, "Status", entity_data["STATUS"])
	node_data.append(create_node_data(status, entity))

	model_change_data = {}
	model_change_data["Operation"] = "Add"
	model_change_data["Minesim"] = True
	model_change_data["Data"] = node_data

	# Create model change event
	evgen = server.get_event_generator(2132)
	evgen.event.Message = ua.LocalizedText(json.dumps(model_change_data))
	evgen.trigger()
```

/Martin