FreeOpcUa/python-opcua #646: List all available variables in server

Tenceto, 2018-07-10:
Hi all,

I was wondering if there is any way to list all the available variables to be measured in a given OPC UA Server. For example, in [OpenOPC](http://openopc.sourceforge.net/api.html) there is an option that lets you flatten out the entire tree into leaf nodes, freeing the user from needing to be concerned with the hierarchical structure.

Is there a way to do this in OpcUa, without needing to search the entire tree for variable nodes?

oroulet, 2018-07-10:
I am having a hard time understanding why someone would need such a thing. It is not easier to search in a flat tree than in a hierarchy... In additions servers are supposed to follow some kind of spec explaining where the interesting variables are

Tenceto, 2018-07-11:
I'm new to OPC and I'm just learning its basics, so maybe you are right and I'm thinking of doing absurd stuff.

I just wanted to generate an array with all the variable names and their paths that can be found in the server. I think this would be useful if the server is not well-documented and the paths to the variables are not specified anywhere. This would let us read some variables automatically rather than searching for them manually using the functions `get_children()` and `get_root_node()`.

Is this too cumbersome?

zerox1212, 2018-07-11:
Doing what you are asking is not impossible, but I don't think it's in the scope of this library. The user could easily make their own recursive search function that collects all the variable nodes to a dict or something.

wattzor, 2019-09-10:
Im also looking for this, to get the whole server three in to a dictonary. How did you solve this ?

oroulet, 2019-09-10:
To state it one more time: if you want to do that you are doing something really wrong. Read and try to understand opcua first

zerox1212, 2019-09-10:
@wattzor You will get more help if you try it yourself first and post your code. Accomplishing this is very basic python. If you don't know how to do it you could try stack overflow first.

wattzor, 2019-09-11:
> 
> 
> @wattzor You will get more help if you try it yourself first and post your code. Accomplishing this is very basic python. If you don't know how to do it you could try stack overflow first.

Well i've done a loop that gets the whole three. But its really slow, and sometimes i get dublicates from .get_childrens()

all im wondering if there is quick way to display whats on a server, the loop methode is really slow , thats why i asked how you solved it. I have a bunch of plcs connected and i just need to know there names and what nodes the plc has

zerox1212, 2019-09-11:
https://github.com/FreeOpcUa/python-opcua/issues/863