FreeOpcUa/python-opcua #647: set moniteringmode failed

chamanth, 2018-07-11:
how to do allow the client to control the monitoring of attributes , when i try to disable it , it comes back to reporting  in UA expert.
please let me know how to give access , is it through "create_monitered_items" , since iam new to this platform ,help in code that needs to be written would be appreciated.
[image]

oroulet, 2018-07-12:
This is probably not implemented in our server yet

chamanth, 2018-07-12:
ok ! 
so what is "create_monitored_items(monitored_items)"  mentioned in the documentation.
if u do not mind would let me know how do use this defintion for our example if is someway related to the what i asked .

thanks in advance!!

chamanth, 2018-07-13:
@zerox1212 @oroulet 

Opc.Ua.ServiceResultException: BadNotImplemented
bei Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
bei Opc.Ua.SessionClient.SetMonitoringMode(RequestHeader requestHeader, UInt32 subscriptionId, MonitoringMode monitoringMode, UInt32Collection monitoredItemIds, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
 bei Opc.Ua.Client.Subscription.SetMonitoringMode(MonitoringMode monitoringMode, IList`1 monitoredItems)
 bei Resources.UASubscription.EnableItems(List`1 items, Boolean enable)

i get the following errors while implementing in opc expert

chamanth, 2018-07-17:
@oroulet @zerox1212 
we are struck in this phase of the project. please help us . without that correcting this, server aggregation is not being possible in opc expert.
[image]
[image]

oroulet, 2018-07-17:
I haven't looked in detailled but most probably disabling reporting for subscriptions is not implemented, so you need to go through the subscription code and add support for it. Probably not very complicated but you need to know well python and read(and understand) the OPC UA spec

zerox1212, 2018-07-17:
Sorry I don't have time to look at implementing this. The best way to get help is to try yourself and make a "work in progress" pull request so we can review it.