FreeOpcUa/python-opcua #648: object.getchild() doesn't work anymore ? "The requested operation has no match to return"

yacinees, 2018-07-13:
Hi,

First, thank you for this amazing package. 
I use it a lot for various projects but recently I encountered a problem while coding a new OPC UA server.
Second, sorry if my english is not great.

When I create my object "MyObject" with the `add_object(index, "MyObject", type)` it creates perfectly my object (after having created its proper type of course).
When I want to add some variables to the type with the `type.add_variable(index, "Variable1", 0` it works also great.

But the problem appears after I have instanciate all of this. When I want to get a child by specifying its name with `my_object.get_child(["%d:Variable1 %index])` it returns "The requested operation has no match to return" like if the child asked doesn't exist ...
This lines of codes worked perfectly for all of my projects since today ...

I have searched in the documentation if there was a change in the package but didn't encountered any response.

If anyone has any idea about this issue, I'll be thankfull !

zerox1212, 2018-07-16:
Does your child have special characters in the browse name?

yacinees, 2018-07-17:
No, the name are really simple like "TEMPERATURE", "PRESSURE" and others parameter like this.

The weird thing is that the same method get_child works perfectly with the client library.
If I get the objects with the `objects = client.get_objects_node()` and then do a `objects.get_child(["%d:TEMPERATURE", %index]` it works really well.

But with the server library it doesn't work anymore ...

zerox1212, 2018-07-17:
Are you using the master branch, or pip?