FreeOpcUa/python-opcua #651: ua.Variant dimensions is an integer but will be packed as array => exception

ephraim, 2018-07-18:
When I try to write a variable which is a list of Int32s, the list get's converted to a Variant with a dimension of 1. This dimensions will later be packed via pack_uatype_array which is wrong as the variant dimension is just an integer.

https://github.com/FreeOpcUa/python-opcua/blob/33c96c801d3a59acdc1825648af53c9346b170a7/opcua/ua/ua_binary.py#L372

This:
b.append(pack_uatype_array(ua.VariantType.Int32, var.Dimensions))

Should be:
b.append(pack_uatype(ua.VariantType.Int32, var.Dimensions))

Right?

Cheers,
Thilo

oroulet, 2018-07-29:
Where do you get an error? In client? Or from server? Can you show the stack?

oroulet, 2018-07-29:
Maybe the error is that we try to save an int but we should save a list of one int