FreeOpcUa/python-opcua #655: How to catch server restart event

Stefan97dk, 2018-07-23:
Hi!

I'm new to Python and Opcua, but I managed to connect to a third party opcua server and read some data.
I also added a datachange_notification for some PLC tags and it works fine.
When I restart my Opcua server the program execution stops with error.
I need to catch the restart event of the server and try connecting again programatically?

Thnx an advance

zerox1212, 2018-07-23:
I don't remember exactly, but I think automatic reconnect is not implemented. You might need a work around that just tries to reconnect ever 20 seconds or something.

Stefan97dk, 2018-07-23:
@zerox1212 thnx alot for ur response.
Indeed, that can be a workaround, but I need to test connection status prior to trying reconnect, any hint how to do this test?

Stefan97dk, 2018-07-25:
Anyone has a hint, would be much appreciated.

oroulet, 2018-07-25:
The only solution, and the one by other clients is to poll the server every second or so, in a thread for example,  and disconnect/reconnect if necessary. Do

Stefan97dk, 2018-07-26:
@oroulet Thank u very much for the hint.
I did try your suggestion by reading a node every second, and I got an exception when I restart the server, and I just catch the exception and try to re-start my application programatically.
The issue is that network event datachange_notification, it doesn't give any exception when the server is been restarted, and the application crashes.

jkbngl, 2020-02-22:
Hi @stefan97dk could you please post your working example? thanks