FreeOpcUa/python-opcua #658: Multi-threaded Subscription

badsmoke, 2018-07-26:
Hello,

I've written a program to subscribe to several thousand variables. But with the initialization of more than 3000 variables it gets so slow that it takes forever.
I would like to start a new thread every 1000 subscriptions, but the subscription function starts its own thread.

I don't know how to communicate the threads.

can someone please explain how I can make this work?

mythread starts subscription -> "subscribtion thread"  subscribe the data-> first thread should write the subscribed data away

thank you very much

oroulet, 2018-07-26:
3000 variables seems a lot.... What kind of server do you have? If you are using the python client, there is only one thread that reads data from socket and one that keeps connection alive. There is not one thread post subscription. I do not see how threading can help you... The issue is clearly parsing fast enough this messages... Python is not good at it... Maybe try with several clients try running each in their own process (process not thread...)

badsmoke, 2018-07-27:
thank you for your answer.

So server we use one PLC, another program subscribes to thousands of data points from this PLC, so it is not the limiting factor.

in the SubHandler description:
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

So he runs in his own thread?

the problem is if I tried my program in several threads or now I've tried it with several processes.
I don't know how to get the data from the subhandler back to my thread/process.

I can show myself via "print" in the SubHandler which he subscribes to and finds the right values, but unfortunately I don't get them transferred to my process anymore.

zerox1212, 2018-07-27:
I am not really sure what your problem is. Are you saying that subscribing to 3000 variables makes your `datachange_notification` call in `SubHandler` slow down?

The call of `datachange_notification` is already from another thread. The comment says don't do expensive operation there because it will block network communications. In `datachange_notification` you should only be storing the new value to your own object/variable. Then you can have another thread do operations on the new data.

badsmoke, 2018-07-27:
oh then i seem to be using the subhandler incorrectly.

```
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """
 
    def datachange_notification(self, node, val, data):
                    for V in range(SubVarCount):
                        if node == SubVar[V][1][P][S][M]: 
                            SubVar[V][0][P][S][M] = round(val,2)
```

Can you send me a code example how to really use the subhandlers to get the data from them?

thanks again

zerox1212, 2018-07-27:
I already made you an example once before in https://github.com/FreeOpcUa/python-opcua/issues/604 .

badsmoke, 2018-07-28:
thanks, i haven't had any contact with classen yet, i will try to understand the code.

I now load all my variables from a file into a variable with multiple lists before each start, so with I only need a general class or?

badsmoke, 2018-07-30:
thanks again, it's all working out so far, I've split it up into several processes now.
currently about 200 data points per process --> total 1000 data points
The PLC can handle this, but my computers ( i5 6200u) are running at 100% capacity on all cores.

In my old thread, it has already been discussed that substription is already the resource-saving variant?