FreeOpcUa/python-opcua #659: RuntimeError: dictionary changed size during iteration

brubbel, 2018-07-26:
Python3, fairly heavy usage and multithreading. PR will follow.

```
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.5/dist-packages/opcua/server/internal_subscription.py", line 289, in _sub_loop
    self.publish_results()
  File "/usr/local/lib/python3.5/dist-packages/opcua/server/internal_subscription.py", line 315, in publish_results
    result = self._pop_publish_result()
  File "/usr/local/lib/python3.5/dist-packages/opcua/server/internal_subscription.py", line 322, in _pop_publish_result
    self._pop_triggered_datachanges(result)
  File "/usr/local/lib/python3.5/dist-packages/opcua/server/internal_subscription.py", line 338, in _pop_triggered_datachanges
    notif.MonitoredItems = [item for sublist in self._triggered_datachanges.values() for item in sublist]
  File "/usr/local/lib/python3.5/dist-packages/opcua/server/internal_subscription.py", line 338, in <listcomp>
    notif.MonitoredItems = [item for sublist in self._triggered_datachanges.values() for item in sublist]
```

oroulet, 2018-07-26:
The trace is missing the error...

closed 2018-07-27 by brubbel

devel0peris, 2018-10-02:
With version 98.5 I still get this error. I'm not iterating over a dictionary in my code. Could this still be the same error?
For privacy reasons, I have replaced project-related traces with `[...]`.

```
ERROR:opcua.common.subscription:Exception calling data change handler
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\common\subscription.py", line 132, in _call_datachange
    self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
  [...]
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\common\node.py", line 578, in delete
    results = opcua.common.manage_nodes.delete_nodes(self.server, [self], recursive, delete_references)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\common\manage_nodes.py", line 390, in delete_nodes
    return server.delete_nodes(params)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\internal_server.py", line 360, in delete_nodes
    return self.iserver.node_mgt_service.delete_nodes(params, self.user)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\address_space.py", line 308, in delete_nodes
    results.append(self._delete_node(item, user))
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\address_space.py", line 325, in _delete_node
    self._delete_node_callbacks(self._aspace[item.NodeId])
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\opcua\server\address_space.py", line 333, in _delete_node_callbacks
    for handle, callback in nodedata.attributes[ua.AttributeIds.Value].datachange_callbacks.items():
RuntimeError: dictionary changed size during iteration
```

devel0peris, 2018-10-08:
> 
> 
> Are you deleting the node inside the `datachange_notification` function?

Yes, I do so. Is it wrong? What is the recommended way to delete nodes within a subscription event?

zerox1212, 2018-10-08:
The way it's implemented right now you probably can't. You will need to mark that node for deletion and remove it from the main thread, not the subscription thread.

devel0peris, 2018-10-09:
Thank you very much. I'm using a workaround now and it works.