FreeOpcUa/python-opcua #662: Trouble writing 32 bit integers

johns10, 2018-07-26:
First off, thanks for all the great work you've done on this module.  I use it almost every day, and have developed a bunch of python scriptlets that accomplish various equipment integration tasks for my company.  It's awesome to work with.

I've solved this with one of your other answers, but I'd really like to understand how to write a 32 bit integer correctly, and hopefully learn a bit in the process.

I'm working with kepware, and the tag I'm working with is a long.  The response from 
```datavalue = tag.get_data_value()
is:
```DataValue(Value:Variant(val:3,type:VariantType.Int32), StatusCode:StatusCode(Goo
d), SourceTimestamp:2018-07-26 21:46:20.195352)
I've got the following code written:
```from opcua import Client, ua
client = Client("opc.tcp://192.168.138.100:49320/")
client.connect()
tag = client.get_node("ns=2;s=FunnelCloud.Simulator.ST05.DowntimeReason")
datavalue = ua.DataValue(ua.Variant(78, ua.VariantType.UInt32))
tag.set_data_value(datavalue)
The response to that code is
```opcua.ua.uaerrors._auto.BadTypeMismatch: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)
However, if I run:
```from opcua import Client, ua
client = Client("opc.tcp://192.168.138.100:49320/")
client.connect()
tag = client.get_node("ns=2;s=FunnelCloud.Simulator.ST05.DowntimeReason")
datavalue = ua.DataValue(ua.Variant(78, ua.VariantType.UInt16))
tag.set_data_value(datavalue)
It succeeds.  The only thing i can figure is that the python integer is an unsigned 16 bit integer (which doesn't seem to add up) out of the box, and it has to match the variant type?  I got it to work from you answer [here]([https://github.com/FreeOpcUa/python-opcua/issues/9]), but I'd really like to understand what's going on.

oroulet, 2018-07-27:
If you read an int32 then you definitely need to write an int32. Does tag.set_value(4, ua.Int32) works? Next step is probably too record a session with Wireshark and see what get read and write. I would be very surprised if we had a bug here

oroulet, 2018-07-27:
What returns tag.get_data_type()?

johns10, 2018-07-27:
Bear with me, I'm a bit green.  
`tag.set_value(4, ua.Int32)`
returns
`AttributeError: module 'opcua.ua' has no attribute 'Int32'`
Do I have a configuration/setup/import I'm missing?  I'm using:
`from opcua import Client, ua`
`tag.get_data_type()`
returns
`TwoByteNodeId(i=5)`
Which would explain why 
`datavalue = ua.DataValue(ua.Variant(78, ua.VariantType.Int16))`
Works.

oroulet, 2018-07-27:
I meant tag.set_value(4, ua.VariantType.Int32)  but now from your variant type it looks like you variable is In16 not Int32 so the strange thing is why the read gives us an Int32. Looks like you need wireshark...

johns10, 2018-07-28:
I'll look at it again on Monday, validate what I'm seeing on quick client and run wireshark.

johns10, 2018-07-28:
Also, I'm really interested in contributing to this project, or to the asyncio branch that cbergmiller is working on.  Is there anything that an amateur/intermediate python programmer can do to help?  I'd even be willing to throw in on documentation or testing, or whatever you guys need.  I'm moving to the asyncio branch for my major project right now, and I'd love to do something to help.  If you want to communicate about it over email, I'm johns10@gmail.com.  I'm no expert, but I'm highly motivated to learn this stuff.

johns10, 2018-07-30:
I've got the async branch running in my app now. It runs with aioamqp for comms to a webserver. I did have issues with several of the examples, specifically with using events. Where should I post bigs for the asyncio version? Here or on the branch? I didn't see an issues tab on the branch. I'll give the bug you referenced a shot. Just bear with me. It's still amatuer hour for me 😀

oroulet, 2018-07-30:
@johns10 just testing, reportingand trying to fix examples and small things is allready plenty of help! I am trying to contact @cbergmiller and I will move/fork his repository to freeopcua group

oroulet, 2018-07-30:
Not from Node. But it is not hard to use the instantiate function

cbergmiller, 2018-07-31:
Hi @johns10, only a few examples have been refactored in the asyncio branch so far.
I just pushed a few commits that contain the updated example for server-events:
https://github.com/cbergmiller/python-opcua/blob/make/asyncio/examples/server-events.py
You can give it a try if you want to.

oroulet, 2018-07-31:
@cbergmiller what about making you a member of freeopcua and moving your repository here?

cbergmiller, 2018-08-01:
@oroulet thanks for adding me to the team! I think it would be a great to move the asyncio version of python-opcua here. I would like to get most of the tests working first though.