FreeOpcUa/python-opcua #666: Cannot connect Ubuntu client to Windows FreeOpcUa Server

pdiaz2, 2018-07-28:
Hello:
My client setup:
- Ubuntu 16.04
- Opcua Package version 0.90.4
- Generated certificates and private key with bash file generate_certificate.sh
- I can run both server and client locally, so opcua is correctly installed in this machine.

My server setup:
- Windows 10
- Opcua package version 0.90.4
- Generated certificates and private key with bash file generate_certificate.sh
- I can run server and client locally. Furthermore, I can connect Matlab OPC UA Client to local server succesfully (this is the reason why the server must remain in Windows host).

Other Info:
- I can ping from both machines in both directions (they are in the same network).
- I can connect FROM a CLIENT in the Windows machine TO the Ubuntu SERVER. 
- This problem still occurs with latest version of opcua package (0.98.3)

Problem:
- I cannot connect a client from Ubuntu machine to a server in the Windows machine. Connection times out with error number 110.

Nothing I have tried out so far can solve the issue.
Thanks in advance

oroulet, 2018-07-29:
this sound like a system/firewall issue nothing to do with python-opcua

pdiaz2, 2018-07-30:
When disabling firewall in Windows host, connection generates error 111: Connection refused.
Do you have any idea/past cases where this situation has happened?

zerox1212, 2018-07-30:
Are you on a corporate network?

pdiaz2, 2018-07-30:
Yes; however, when checking for other imposed firewalls through:
netstat -ano | findstr -i SYN_SENT

nothing is returned, so I am assuming there are no other firewalls active (not forgeting the fact that I can connect with a client from Windows to server on Ubuntu)

closed 2018-08-17 by pdiaz2