FreeOpcUa/python-opcua #669: contradiction with the specification document

Hootan-Man, 2018-07-30:
https://github.com/FreeOpcUa/python-opcua/blob/10640f4e81eda5dad3ab769abc5ce62dc8d4b93f/opcua/client/client.py#L314

sorry, but acoording to "OPC Unified Architecture Specification Part 4: Services"

> 5.4.5 RegisterServer
> 5.4.5.1 Description
> This Service is implemented by Discovery Servers.
> This Service registers a Server with a Discovery Server. This Service will be called by a Server or
> a separate configuration utility. Clients will not use this Service.**

why you are calling this service on the client side?
Am I missing a point ?!

Hootan-Man, 2018-07-30:
1st I agree that the documentation is not well written and could have been much better that what it is today.
let me describe it in this way: 
we have an entity which called "discovery server" we can think of it as a service broker.
we have "server" as a service provider. it is a duty of service provider (server) to tell to the broker (Discovery-server) where and what kind of services it provides and what are the policies.
on the other side the service consumer (client) should only ask the broker where he can find the service he is interested in. 
in real life (as consumer) you can do that and tell to a broker about a service which he never heard  of it. but in software world if we do not follow the good practices that Object/service Orientation design recommends at the end we make mess. it is the duty of the server to inform the Discovery-Server about himself. and client should not intervene.

oroulet, 2018-07-30:
I perfectly know these concepts. But fom the server you still need a client to inform the discovery server about your services. We could have made a specialized client for that, but it is still a client

Hootan-Man, 2018-07-30:
sorry, I didn't mean that you don't know about it, but it happens to every one to make a mistake(including me at this moment).
I'm sure you guys are much better than me, since it's couple of months that I started to read about OPC-UA and it's years that you have your hands on coding.
I just read the docs, and I was reading some sample codes to see how different developers with different languages implemented the protocol.
here is the reference to another implementation
https://github.com/open62541/open62541/tree/master/src/client
if you look at their code they don't call this service on the client side.

I just don't understand why we need a client for registration?!!! 
can you explain it in a more simple way. (I haven't read all your codes yet)

Hootan-Man, 2018-07-31:
yes you are right.
for talking to a server you need to be a client.
but according to the documentation:

> If a Server acts as a Client to other Servers, which is commonly referred to as Server chaining,

and it is not the only scenario that we have so called "server chaining".
think of it as the case that you are using a website, which uses a web service to respond you.
 
you are providing this service to every client to register in the discovery server.
which may be abused by some clients to divert the traffic, or in some other sense (for example, sending a lot of request may result in a DDoS attack).
what is obvious is that, client has an extra behavior which he shouldn't have.

zerox1212, 2018-07-31:
You might be correct. However moving this to the server code wouldn't solve the problem. This library has both a client and a server. And either can be created with a few lines of code. I think you could abuse this regardless just because of the design of discovery.