FreeOpcUa/python-opcua #671: Python3.7 --->  SyntaxError: invalid syntax--->version 0.98.3

sehb, 2018-08-06:
E:\python\code\PythonUAClient\venv\Scripts\python.exe E:/python/code/PythonUAClient/client_to_prosys.py
Traceback (most recent call last):
  File "E:/python/code/PythonUAClient/client_to_prosys.py", line 5, in <module>
    from opcua import Client
  File "E:\python\Python37\lib\site-packages\opcua\__init__.py", line 5, in <module>
    from opcua.common.node import Node
  File "E:\python\Python37\lib\site-packages\opcua\common\node.py", line 6, in <module>
    from opcua import ua
  File "E:\python\Python37\lib\site-packages\opcua\ua\__init__.py", line 7, in <module>
    from opcua.ua.uaprotocol_hand import *
  File "E:\python\Python37\lib\site-packages\opcua\ua\uaprotocol_hand.py", line 5, in <module>
    from opcua.common import utils
  File "E:\python\Python37\lib\site-packages\opcua\common\utils.py", line 171
    task = asyncio.async(coro, loop=self.loop) 
                       ^
SyntaxError: invalid syntax

zymergen-jcoats, 2018-08-07:
It looks like https://github.com/FreeOpcUa/python-opcua/pull/649 addresses this problem by changing `asyncio.async` to `asyncio.ensure_future`, but it didn’t make it into the 0.98.3 tag. @oroulet, is there a possibility of us getting a new patch release?

sehb, 2018-08-08:
Yes,it works
Thanks @zymergen-jcoats

closed 2018-08-08 by sehb