FreeOpcUa/python-opcua #673: Help: Performing writes outside of event handler

acarmstead, 2018-08-07:
Hello, forgive me as I am new to Python.

I am subscribed to a handful of tags on Kepware, and have implemented the datachange_notification function incorrectly because I am trying to make writes to tags depending on the value I receive, but this is not allowed by python-opcua as far as I can tell.

Is there a way to work around this? I was thinking a Queue but I have no experience with multiprocessing

acarmstead, 2018-08-07:
Can anyone point me in the right direction on starting a separate thread that can call set_value after I receive an update in datachange_notification?

oroulet, 2018-08-08:
Read any threading tutorial is the best I suppose

zerox1212, 2018-08-08:
It's a more advanced approach, but if you are storing your values in your own object you could do some reading on overriding the built in `set_attribute'. Then every time your variable gets updated by OPC UA you could detect it and trigger something from the main thread (or another thread). This would be an event based approach instead of a timed approach, but it's getting into some advanced Python knowledge.