FreeOpcUa/python-opcua #680: Help: using Siemens WinCC to access python server

ElisabethMZ, 2018-08-16:
I am having difficulty accessing an opcua server created in python using WinCC.
Has anybody had success getting a connection to work?

Siemens support insists the problem must be caused by improper handling of the certificate in python-opcua

Thank you for your support!

oroulet, 2018-08-16:
we need more info, what problem do you have? what error?

ElisabethMZ, 2018-08-16:
FYI: Other OPC UA clients can access the Python OPC UA server and the WinCC client can access other servers. Only the combination Python OPC UA Server and WinCC Client does not work.

I tried three different ways:
1. create the Python server without "load_certificate" and without "load_private_key
Security settings in WinCC Client: None / None

Result: Variables cannot be browsed, no connection to server
Error message in WinCC: "No certificate used by the server to establish secure access".

2. create the Python server with "load_certificate" and with "load_private_key
Security settings in WinCC Client: None / None

Result: Variables cannot be browsed, no connection to the server
Error message in WinCC: "The resolution of the IP address failed. Use the computer name instead." (When using the computer name instead of the IP address, the same error message appears)

3. create the Python server with "load_certificate" and with "load_private_key
Security settings in WinCC Client: Basic128Rsa15 / Sign and encrypt

Result: Variables can be browsed, no communication with the server during runtime, no error message

zerox1212, 2018-08-16:
You will probably have to use wire shark to see what WinCC is really doing.

oroulet, 2018-08-16:
Yes you will need to record a session with Wireshark in case 2 so we understand what they want. But no certificate is necessary when selecting None/None. This is a most probably a bug on their side

zerox1212, 2018-08-16:
I think someone on here once posted that WinCC requires encryption by default.

ElisabethMZ, 2018-08-17:
I am currently in contact with the Siemens Support and will keep you updated.
Thank you for your help :)

ElisabethMZ, 2018-08-17:
The following statement came from Siemens Support:
“My level of knowledge is that even with a None-None configuration, a certificate with "no encryption" is transmitted and therefore has less security requirements.
There are guidelines from the Foundation pages, which we are quite sure to follow. We are very sensitive about this, which means that if there is something that is not where it should be, then we break off the connection.
So I don't think WinCC is the cause.”

oroulet, 2018-08-17:
I had a discussion with the product people a few month ago on that topic and they were confident that a certificate was not necessary. But anyway, they can require a certificate if they want and inform users. But the second scenario need to be debugged. Can you save a session with Wireshark or similar and attach to this bug report?

oroulet, 2018-08-17:
Record one session with Python and one with another client that works

brubbel, 2018-08-29:
@oroulet Could this be related to the uaprocessor replying to FindServersRequest with the default endpoint-url of 'opc.tcp://**0.0.0.0**:4840/freeopcua/server/'?
I see that UaExpert also gives a remark on this, then overrides the 0.0.0.0 with the url which was used to contact the local discovery service, which then works.

But some clients won't do that.
Also, I did not find anything in the opc-ua standard that mentions this 0.0.0.0 (which is in fact the bind-ip of the server).
Is there any objection against detecting this 0.0.0.0 and replacing it by the actual endpoint-url which is available in the client-request?

The above would allow binding to multiple interfaces and still use clients that are confused by this 0.0.0.0 (and even allow vpn and/or use port forwarding without UA-GDS).

References:
http://documentation.unified-automation.com/uasdkdotnet/2.5.3/html/classUnifiedAutomation_1_1UaBase_1_1FindServersRequest.html

brubbel, 2018-08-30:
@ElisabethMZ Could you provide feedback if the following condition works for WinCC:

1. Create freeopcua server with load_certificate(), although not used.
2. Also: [server.set_endpoint('opc.tcp://x.x.x.x:4840/freeopcua/server/')](https://github.com/FreeOpcUa/freeopcua/blob/master/python/README.md), where x.x.x.x is the ip-address of the interface of the machine you are running the server.
3. Connect WinCC client using Security = None

ElisabethMZ, 2018-09-04:
Thank you for your comments.

Now I have new experiences:
1. My WinCC client also does not connect to other servers if security is active (not even with a Siemens PLC). That's why I couldn't use Whireshark to make any comparisons.
I am currently waiting for a reply from Siemens.

2. load_certificate() does not work. This error message appears: TypeError: load_certificate() missing 1 required positional argument: 'path'

brubbel, 2018-09-06:
@ElisabethMZ 
[This branch](https://github.com/brubbel/python-opcua/tree/wincc) works. I have tested it with TIA + WinCC.
The correct settings are as follows:
1. Start freeopcua with load_certificate('path_to_cert.pem'). Verify with UaExpert if certificate works.
2. Set WinCC to use Basic128Rsa15, Sign & encrypt.
3. Move the certificate from 'rejected' folder to 'certs' as explained in [this document](https://cache.industry.siemens.com/dl/files/236/63481236/att_917852/v2/63481236_Part5_RT_Advanced_Server_und_Panel_Client_en.pdf) section 3.3.

Apply all steps carefully. WinCC is quite picky on the application of the opc-ua standard.

ElisabethMZ, 2018-09-07:
@brubbel what version of WinCC 15 are you using? We could not get it working with WinCC Professional.
The official OPC Scout also developed by Siemens can access everything required.

Just heard back from the support, they are passing our channel diagnosis data on to the development team 👍
Thank you for your support thus far.

brubbel, 2018-09-07:
[WinCC Advanced Version V15](https://support.industry.siemens.com/cs/document/109752566/simatic-step-7-and-wincc-v15-trial-download) it says.

ElisabethMZ, 2018-10-02:
Just heard back from the Siemens support a fix is coming in V15 SP1.

oroulet, 2018-10-02:
I like it when, with our small part time hobby project, we trigger bugs in industrial software installed all around the world...