FreeOpcUa/python-opcua #682: ImportError('No module named type_dictionary_buider',)

alexrdle, 2018-08-24:
With the newest changes in this project the custom structure example doesent seem to be working anymore.
https://github.com/FreeOpcUa/python-opcua/blob/6fb007212417c28652a8e4d2e1b9f753a8ef45ae/examples/server-create-custom-structures.py

oroulet, 2018-08-24:
@yangpeiren

yangpeiren, 2018-08-24:
@alexrdle since the new module opcua.common.type_dictionary_buider.py is not released in the pypy package opcua (@oroulet , I've just tested with a clean computer), if you cloned the project to local, please make sure your program imports modules from your cloning folder, one possible way to test is that you can use pip to uninstall the opcua and try to run any examples from the example folder, if no opcua is found, then your program imports modules from site-packages, not from local cloned repository.

alexrdle, 2018-08-27:
@yangpeiren thanks for the answer. When will there be a new release candidate for the pypi opcua package?

yangpeiren, 2018-08-27:
@oroulet

alexrdle, 2018-08-27:
@oroulet  @yangpeiren okey, thanks guys! Closed

closed 2018-08-27 by alexrdle