FreeOpcUa/python-opcua #684: TimeoutError : Multiple OPC Servers support from singe Client

satheesh90, 2018-08-27:
I'm writing a client code in python to access 2 OPC Servers simultaneously and subscribe to the interested **tags data values** in the OPC servers. My objective is to access and subscribe to two different OPC servers using **threading** or **multiprocessing**. (Note: OPC Servers I mention here is nothing but the KEPWARE Simulator running in two different machines in the same network where the client code runs in an IoT Gateway).

I face an **Error** after commencement of client code execution in around 45 mins to 1 hour as below:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "../opcua/client/ua_client.py", line 94, in _run
    self._receive()
  File "../opcua/client/ua_client.py", line 103, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "../opcua/ua/uaprotocol_hand.py", line 671, in receive_from_socket
    return self.receive_from_header_and_body(header, utils.Buffer(body))
  File "../opcua/ua/uaprotocol_hand.py", line 642, in receive_from_header_and_body
    header, body)
  File "../opcua/ua/uaprotocol_hand.py", line 392, in from_header_and_body
    crypto.verify(obj.MessageHeader.to_binary() + obj.SecurityHeader.to_binary() + decrypted, signature)
  File "../opcua/crypto/security_policies.py", line 155, in verify
    self.Verifier.verify(data, sig)
  File "../opcua/crypto/security_policies.py", line 276, in verify
    raise uacrypto.InvalidSignature
InvalidSignature

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "../opcua/client/client.py", line 62, in run
    val = server_state.get_value()
  File "../opcua/common/node.py", line 129, in get_value
    result = self.get_data_value()
  File "../opcua/common/node.py", line 138, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "../opcua/common/node.py", line 244, in get_attribute
    result = self.server.read(params)
  File "../opcua/client/ua_client.py", line 293, in read
    data = self._uasocket.send_request(request)
  File "../opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/concurrent/futures/_base.py", line 464, in result
    raise TimeoutError()
TimeoutError

After this error occurs, the Thread accessing one OPC server is stopped and the second Thread accessing second Server continues its operations for some more time (~2 to 10 mins). After that, second thread also fails displaying the same error. 
Suggest me a way to solve this or some other better way to implement multiple server support from a Single client code. Thanks

oroulet, 2018-08-29:
looks similar to this one https://github.com/FreeOpcUa/python-opcua/issues/652
There is obviously need for some improvment in that code...

oroulet, 2018-08-29:
Can you try: #686

satheesh90, 2018-08-29:
Hi @oroulet : I tried running my client code after updating the changes you committed. Now, the InvalidSignature error is gone. Still, TimeoutError is persistent!! I get as below after some 20 mins running the Client code:

Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "../opcua/client/client.py", line 62, in run
    val = server_state.get_value()
  File "../opcua/common/node.py", line 129, in get_value
    result = self.get_data_value()
  File "../opcua/common/node.py", line 138, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "../opcua/common/node.py", line 244, in get_attribute
    result = self.server.read(params)
  File "../opcua/client/ua_client.py", line 293, in read
    data = self._uasocket.send_request(request)
  File "../opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/concurrent/futures/_base.py", line 464, in result
    raise TimeoutError()
TimeoutError

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "../opcua/client/client.py", line 62, in run
    val = server_state.get_value()
  File "../opcua/common/node.py", line 129, in get_value
    result = self.get_data_value()
  File "../opcua/common/node.py", line 138, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "../opcua/common/node.py", line 244, in get_attribute
    result = self.server.read(params)
  File "../opcua/client/ua_client.py", line 293, in read
    data = self._uasocket.send_request(request)
  File "../opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/concurrent/futures/_base.py", line 464, in result
    raise TimeoutError()
TimeoutError

oroulet, 2018-08-29:
don't you get a waring about invalid signature before?

but yes you will get a timeout error. I fixed the bad handling of exception but there is still a bug in handling of signature.. I never looked at the encryption code but if the error happens after such a long time it might be that we should update signature after some special message from server... not sure