FreeOpcUa/python-opcua #685: ``ServerTimestamp`` now missing in datavalue

nepix32, 2018-08-29:
Hello,

[this commit](https://github.com/FreeOpcUa/python-opcua/commit/135159e594ed04ff67a08fd31901bd78bf4a668b) removed the availability of ``ServerTimestamp`` attribute in a datavalue. I am asking because I have a client here, which relies on the ``ServerTimestamp``. 

In the commented out section ``ServerTimestamp`` has been added up until the commit above.

     def set_attribute_value(self, nodeid, attr, value):
            ...
            if attr not in node.attributes:
                return ua.StatusCode(ua.StatusCodes.BadAttributeIdInvalid)
            #if not value.SourceTimestamp:
            #    value.SourceTimestamp = datetime.utcnow()
            #if not value.ServerTimestamp:
            #    value.ServerTimestamp = datetime.utcnow()

In ```Node.set_value`` only the ``SourceTimestamp`` is added and ``ServerTimestamp`` is not there any more.
 
     def set_value(self, value, varianttype=None):
        ....
         datavalue = None
         if isinstance(value, ua.DataValue):
            datavalue = value
        elif isinstance(value, ua.Variant):
            datavalue = ua.DataValue(value)
            datavalue.SourceTimestamp = datetime.utcnow()
        else:
            datavalue = ua.DataValue(ua.Variant(value, varianttype))
            datavalue.SourceTimestamp = datetime.utcnow()

Any advise how to get that functionality back? My woraround is to  ``pip install opcua<=0.98.3``, which will preclude any further improvements for me.

oroulet, 2018-08-29:
What was removed is forcing the presence of Timestamps. But SourceTimestamp is created when you write to a variable the normal way. You can also, as before, control the entire process byt writting a DataValue object with Node.set_data_value(dv). You can then set whatever timestamps you want

But what kind of client want ServerTimestamp? ServerTimestamp is not present in most servers I have seen

nepix32, 2018-08-29:
Hello, for clarification:

I only get my hands on a few types of OPC-UA-Servers:
Copadata Zenon process gateway
Siemens OPC-UA (Simatic.S7, WinCC and T3000)

All of them implement SourceTimestamp and ServerTimestamp on a datavalue

As said, I do not require a change in the library. But could there be advice about monkeypatching or something the like?

oroulet, 2018-08-29:
You can always fork and change these lines. By I would like to know what client is giving issues with this. What client? And what error?

nepix32, 2018-08-29:
It is an in-house client which relies on the ``ServerTimestamp`` (do not ask why, but it breaks for me since the commit in question. I use opcua library to make a "simple test server".

I want to avoid forking, I would more be interested in monkey-patching.

nepix32, 2018-08-30:
To nudge a little bit more:

I would never have noticed the issue with our internal client software had i not seen that in the current version of python-opcua the ``ServerTimestamp`` looked as follows in UAExpert. If it had not been so prominently been displayed in UAExpert I would never had figured out what my problem was. And at least the sever timestamp seems to be important enough for the programmers in UAExpert to put it in the tag display.

[image]

I make the (unfounded) statement that other free OPC-UA implementations add the ``ServerTimestamp`` too (like node-opcua).

But enough of that, for me it is ok, maybe someone else has an idea how to monkey-patch or I'll find out on my own.

oroulet, 2018-08-30:
Is this an issue the nodes in address space or for you custom nodes? If this is for your custom nodes then you just need to write that server timestamp when writting values: , there is an example here: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-example.py#L159 just set ServerTimestamp on the DataValue before writting it

nepix32, 2018-08-30:
Thanks for the advice, tried the following:

        dv = ua.DataValue(value)
        dv.ServerTimestamp = datetime.utcnow()
        if time_stamp is None:
            time_stamp = dv.ServerTimestamp
        dv.SourceTimestamp = time_stamp
        self.server.set_attribute_value(opc_obj.nodeid, dv)

Now I can set the source/server timestamp to whatever I want.

OK for me to close issue.

shamalb, 2019-05-29:
Hello. 
We have got the same situation: we made opcua server based on freeopcua.
In our case we used for test KEPware OPC-bridge OPCDA2OPCUA and we don`t have the sourcetimestamp of values on opcua server from timestamp of values on OPCDA server. Also we tested it with UAExpert and we have the same question: how repair it?

cncgeorge, 2019-10-15:
Team,
I can use some help, I am working where the requirement for the server timestamp is a must and I don't how to add it. Here is my server.py file. Any suggestions would be greatly appreciated.
# get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()
    print('objects',objects)
# populating our address space
    myobj = objects.add_object(idx, "TendAI Objects")
    myobj.ServerTimestamp = datetime.datetime.now()
    print('myobj',myobj)
    myprop = myobj.add_property(idx, "myproperty", "I am a property")
    print(myprop)
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()    # Set MyVariable to be writable by clients
    #myvar.DataValue.ServerTimestamp(datetime.datetime.now())

[server.txt](https://github.com/FreeOpcUa/python-opcua/files/3727058/server.txt)

sosi-deadeye, 2020-10-20:
Actually, it's problematic to use python-opcua with integrated OPC UA Server on SIEMENS PLCs.

I don't know why, but writing a SourceTimestamp is not allowed. I forked the project, made a small change to fix it.
https://github.com/sosi-deadeye/python-opcua/commit/570977503a333998f84bd26b752456f7b4fbd807

We should have control over all internal default attributes of the Client class and Node class.
For example, the standard setting for a session_timeout on a SIEMENS PLC is 30 seconds.
The internal default Value for session_timeout in the class Client is one hour.