FreeOpcUa/python-opcua #689: secure connection between server and client

RasPiTool, 2018-09-04:
Hello folks
i would like to build any secure connection between server and client. The server should verify client authentifikation before I can use methodes like client.get_root_node() or accessing to objects on server side.
I tried it with certificate and private-key as in examples, but it doesn't work. The server and client can have different certificates and the connection can be established. 

I tried to biuld a secure connection that way, but it doesn't work also.
```

server.set_endpoint("opc.tcp://PI01:Raspi@192.164.0.119:4840/freeopcua/server/")

client = Client("opc.tcp://PI02:Raspi2@192.168.0.119:4840/freeopcua/server/")

```
The username and password are different on both sides and the connection can be established.

My next approach is use the methodes like 
```

server.set_security_policy()
server.set_security_IDs()

client.set_security()
client.set_security_string()

```
I tried to use this methodes and I don't get it, how it works.

Can somebody help me?

I would be realy happy!

oroulet, 2018-09-04:
The server and clients were designed to connect to as many several/client as possible... Not to be secure... But maybe it is time to change that

RasPiTool, 2018-09-04:
Thanks for answer!

closed 2018-09-04 by RasPiTool

zerox1212, 2018-09-04:
If you are an experienced programmer you should be able to improve the security. However the crypto/cert stuff is rather specialized so if you are a new programmer it will be difficult.

I thought the basic cert system was working to the point where it would at least reject the connection. Maybe it was that only encryption over the wire was working.

RasPiTool, 2018-09-07:
thanks for answer,
I am new in programming. I am working on my master thesis and i'm student at faculty mechanical engineering. I think that's enough for me.