FreeOpcUa/python-opcua #699: Question: Slow first raw_history_read ?

track02, 2018-09-18:
Hello,

Firstly thanks for your work on this library I've found it really useful.

Just a quick question I'm hoping you could help with:

I have a python-opcua server but I'm working with a different client library that does not currently support historical reads.

I've tried to make a temporary work-around by exposing a function on the server that takes in two dates and returns a csv string of node data within that interval. (These are fairly small intervals)

Basically it's just a call to:

`my_historical_node.read_raw_history(start_date, end_date)`

This works fine however after starting the server the first request to this function is very slow (30-60s) but any subsequent calls are pretty much instant.

Is this to be expected or am I just making a stupid mistake?

Many thanks

oroulet, 2018-09-18:
You are making a stupide mistake ;-)

oroulet, 2018-09-18:
I have at work servers supporting several clients calling methods  at 200hz and they still answer

track02, 2018-09-18:
Could I ask how you would do it? Is using `read_raw_history` not the best way?

zerox1212, 2018-09-18:
What history backend is your server using? In memory or SQLite?

oroulet, 2018-09-18:
Can you try using cprofile module to profile your program? Just start server, run your method then close. Then you will see what method is taking so long

track02, 2018-09-18:
I'll give that a try in the morning and let you know if I find anything of interest - thank you both.

track02, 2018-09-23:
Apologies for the late reply.
I did have a look into this, it wasn't `read_raw_history` that was slow but rather the processing of its returned results. 

Instead I'd like to just store the returned history list in a variable and have the client subscribe and handle it as needed, so I've made a new variable and update it with the history, a bit like this:
```
result = self.vars.add_variable("ns=2;s=HistoryResult", "History Result", [], ua.VariantType.DataValue)
...
result.set_value(read_node.read_raw_history(start, end))
```

This seems fine server side, no hold up anymore but I get a conversion warning on the client when it detects a change in value:
`qt.opcua.plugins.open62541: Variant conversion from Open62541 for typeIndex 22  not implemented`

Before I start looking further into Open62541 could you confirm this the correct way to setup the result node for the history list? And if so, please feel free to close the issue.

Thanks again

closed 2018-09-28 by track02