FreeOpcUa/python-opcua #701: Best Practice For Handling Multiple Subscriptions

bryankim96, 2018-09-20:
Hi, I'm trying to mirror OPC UA objects with Python objects on the client side, similar to the approach shown in https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-ua-python-mirror.py (except on the client side) and discussed in issue #502. I carefully examined both the example and the issue, but I'm not sure what the best approach is to client-side mirroring because of this network-blocking issue which was discussed in some detail in the issue. 

The main problem I have is that I want to be able to connect to an OPC UA server, mirror some objects w/ Python objects, subscribe to their variables/properties, and, receive data change notifications which uniquely identify which node's value was changed. Essentially my question is the same as @zerox1212 's here:

> can you post a better design for subhandler? I can't see a clear way to use SubHandler inside an arbitrary python class. You always have to look up what node has called `datachange_notification`. I'm still confused on how best to use this design...

I saw @zerox1212 's suggestion with caching the node name in https://github.com/FreeOpcUa/python-opcua/issues/502#issuecomment-338756695. However, as far as I can tell this solution is only suited for monitoring individual nodes. If I wanted to make a Python class which corresponds to a UA object (like was done in the server-ua-python-mirror.py example), and I want it to own a single subscription to its child property/variable nodes, there doesn't seem to be any way within a SubscriptionHandler's datachange_notification() to actually see which node was changed when a data change notification is given...

One way I can see for handling this is to basically duplicate the single-node node name caching method for every node, but as far as I can tell this would require making a separate subscription for each node, which from what I can see from the documentation of other OPC UA implementations, seems like a really bad idea. Is there something I'm missing? @oroulet, is there somewhere I can find a good pattern for this use case? Maybe using some of the lower-level functionality would help?

Thanks in advance.

oroulet, 2018-09-20:
I never really understood the issue but it is usually more efficient to have few subscriptoins. In the callback you get the nodeid so if you have a dict of nodeid->atributes somewhere is should be very easy and efficient to update all your nodes:

for example in something siilar to that: (not tested hypotethical code)

db = {}
myobject # an object with variable variables
for attr in myobject.__dict__.keys():
     node = my_parent_node.add_variable(attr, getattr(myobject, attr))
     sub.subcrbe_data_change(node)
     db[node] = attr

then in handler
    def callback_data_change(nodeide, val, data):
        setattr(myobject, db[nodeid], val)

zerox1212, 2018-09-20:
At the moment you have to do what @oroulet says. You must keep track of every subscribed node then do a look up of it's corresponding python attribute in `datachange_notification`.

While the above solution is simple in practice I think the design could be better. I would like to try making a more traditional "call back" design some day where adding a node to a subscription automatically keeps track of the link. In otherwords, when adding a node to the subscription you would also supply your callback function. This function could be method of your class instance. That way a data change of the supplied node would automatically call the "call back" method you provided in advance. Not sure how to get the instance method to be called on the main thread, but that would be idea.

But for now we have this static function being called that needs a look up.

zerox1212, 2018-09-20:
Thinking about it more there should be a more elegant way to make the mirror on the client without any changes. If I get some time in the next few days I'll will try to create an example.

bryankim96, 2018-09-20:
Ok, thanks for the suggestions! For now I'll do as @oroulet suggests and use the nodes as keys.

Your suggestion regarding callbacks sounds like a good one to me. It would certainly be nice to have custom behavior for different nodes rather than having to set everything at the subscription level (especially when it seems like the recommended design pattern is to have only 1 or a few subscriptions each handling many monitored nodes).

I'm not sure I have the time (or skills) to make the improvement you propose, but I'll try to make an attempt sometime if I can.

zerox1212, 2018-09-20:
I also didn't realize having many subscriptions was a problem. The server UA mirror example definitely doesn't follow this convention because it creates a subscription for every object. On the other had my use case never had a problem and I had several hundred objects (thus several hundred subscriptions).

oroulet, 2018-09-25:
The library was design to monitor the way subscriptions are defined in the spec. It is possible to write more code to have one subscription per node, but the result will not be more efficient than doing it yourself.

btw I was wrong about performance. Early we used one thread per subscription so it was quite heavy, now we use asyncio and having several subscriptions probably does not cost much. But it would be interesting to test it for extreme cases. 10000 subscriptions vs 10000 nodes in one subcrtions,,,