FreeOpcUa/python-opcua #705: referencing between two nodes using Expanded Node ID

akgaurihar, 2018-09-24:
Hello,
I am trying to create a reference from one node in one server to the another node in another server using NodeID, but it is not working, it is possible using Expanded Node ID it seems, please help me to implement this function. how should I use Expanded Node ID for creating a reference?
Thanks in advance!

oroulet, 2018-09-24:
In theory this is supported but absolutely not tested and probably missing some easy api. Every node has a namespace uri. Try to set it to somehting to get started..

akgaurihar, 2018-09-24:
as of now I tried using get_node to create a link between two nodes, it worked, but this is not the proper implementation for creating reference.

oroulet, 2018-09-24:
What do you expect?

oroulet, 2018-09-24:
Try to look at code of add_object for example. Copy it and modify to create an extended nodeid... Should be possible...

akgaurihar, 2018-09-24:
Okay, I will let you know what happens! thank you:)

Molokay, 2018-10-30:
Did you get this to work ? I am especially interested if you could use UAExpert to follow the references across servers. I have the feeling neither UAExpert nor the free opcua-client can follow references across servers right now.