FreeOpcUa/python-opcua #706: Get all available nodes

pinkynrg, 2018-09-24:
I'm writing a driver to easily read/write tags. 
I'm also implementing a search function to search a node among the available tags.
I find all available node id by recursively using the get_children method. 
The recursive method I'm currently using though is kind of slow since I have many tags. 

Is there a way to get a list of all available node ids without using the recursive method explained above?

zerox1212, 2018-09-24:
This has been discussed in https://github.com/FreeOpcUa/python-opcua/issues/646. 

There is nothing built in to the library to do this.

pinkynrg, 2018-09-24:
fantastic thanks! closing

closed 2018-09-24 by pinkynrg

zerox1212, 2018-09-24:
I suppose the "OPC UA" answer to this question could be that the server should create `Views` of variable sets. However Views are not implemented.