FreeOpcUa/python-opcua #707: NodeClass: Invalid Datatype

ibrahimsadek, 2018-09-25:
I am running a simple opc-ua server with one object and three variables as indicated below. I used the uaExpert as a client. According to uaExpert, the NodeClass for either the object or the variables is invalid and there error is (see attached) : "Invalid Datatype: Expected OpcUaType_Int32, Received OpcUaType_UInt32", nevertheless, the uaExpert is working properly and I can see the values. 

Can someone kindly explain to me how to define this NodeClass as  "OpcUaType_Int32" instead of "OpcUaType_UInt32" ?

[image]

import datetime
import time
from random import randint

from opcua import Server, ua

server = Server()

url = "opc.tcp://my_IP:4840"
server.set_endpoint(url)

server.set_server_name("FreeOpcUa Example Server")

server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt])

server.load_certificate("my_cert.der")
server.load_private_key("my_private_key.pem")

name = "OPC_SIMULATION_SERVER"
idx = server.register_namespace(name)

objects = server.get_objects_node()  # root node

myobj = objects.add_object(idx, "MyObject",0)
Temp = myobj.add_variable(idx, "Temperature",0)
Press = myobj.add_variable(idx, "Pressure",0)
Time = myobj.add_variable(idx, "Time",0)

Temp.set_writable()
Press.set_writable()
Time.set_writable()

server.start()
print("Server started at {}".format(url))

try:

    while True:
        Temperature = randint(10, 50)
        Pressure = randint(200, 999)
        TIME = datetime.datetime.now()

        Temp.set_value(Temperature)
        Press.set_value(Pressure)
        Time.set_value(TIME)

        time.sleep(2)

except KeyboardInterrupt:
    pass

finally:
    server.stop()
    print('Server Disconnected')

ibrahimsadek, 2018-09-25:
This issue occurred when the opc-ua is installed via git clone (Master). However, when I used pip install (released version) it does not exist. On the other hand, the released version does not support "Basic256Sha256_SignAndEncrypt" security policy. Shall this option be enabled in the released version ? as I have to use a different security option.

Thanks.

oroulet, 2018-09-25:
I reverted the change, it should work now

closed 2018-09-30 by ibrahimsadek