FreeOpcUa/python-opcua #708: BadWriteNotSupported error with latest version (0.98.5) only

fcep, 2018-09-25:
Dear all, I have an issue with the latest version of python-opcua. All versions up to 0.98.3 work fine, but with version 0.98.5 I get the following error:
```
Traceback (most recent call last):
  File "./client_example_minimal.py", line 23, in <module>
    var.set_value(True) # set node value using implicit data type
  File "/usr/lib/python3.7/site-packages/opcua/common/node.py", line 211, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/usr/lib/python3.7/site-packages/opcua/common/node.py", line 257, in set_attribute
    result[0].check()
  File "/usr/lib/python3.7/site-packages/opcua/ua/uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)
```

This is the code that generates the error (taken by the client-minimal example):
```python3
#!/usr/bin/python3

import sys
sys.path.insert(0, "..")

from opcua import Client

if __name__ == "__main__":

    client = Client("opc.tcp://localhost:4840")
    try:
        client.connect()

        # get a specific node knowing its node id
        var = client.get_node("ns=4;s=MAIN.OutletTLSC.CloseCMD")
        #var = client.get_node("ns=4;s=MAIN.ConditionAutoClose.sGiveHour")

        #print(var)
        #var.get_data_value() # get value of node as a DataValue object
        #var.get_value() # get value of node as a python builtin

        #var.set_value(ua.Variant([23], ua.VariantType.Int64)) #set node value using explicit data type
        var.set_value(True) # set node value using implicit data type

    finally:
        client.disconnect()
```
I hope you can help me run the latest version of the software.
Thank you.

fcep, 2018-09-26:
Hi, I tried doing:
```
        dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
        dv.ServerTimestamp = datetime.now()
        dv.SourceTimestamp = datetime.now()
        var.set_data_value(dv)
```

But I get exactly the same error as before:
```
Traceback (most recent call last):
  File "./client_example_minimal.py", line 29, in <module>
    var.set_data_value(dv) # set node value using implicit data type
  File "/usr/lib/python3.7/site-packages/opcua/common/node.py", line 211, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/usr/lib/python3.7/site-packages/opcua/common/node.py", line 257, in set_attribute
    result[0].check()
  File "/usr/lib/python3.7/site-packages/opcua/ua/uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)
```

Is my code correct?
Thank you.

oroulet, 2018-09-26:
Try different combinations

fcep, 2018-09-26:
I tried, but nothing seems to work.
As a side note, the command line tool uawrite works fine, for example with the following:
`uawrite -u "opc.tcp://localhost:4840" -n "ns=4;s=MAIN.OutletTLSC.CloseCMD" -t guess True`

But I need the code to work inside a Python script.
Any ideas? Thank you.

fcep, 2018-09-26:
Unfortunately on my machine I cannot even run the old versions of the software anymore, as with every version except the latest it says
```
File "/usr/lib/python3.7/site-packages/opcua-0.98.3-py3.7.egg/opcua/common/utils.py", line 171
    task = asyncio.async(coro, loop=self.loop) 
                       ^
SyntaxError: invalid syntax
```

fcep, 2018-09-26:
I managed to run version 0.98.3 by applying:
```
--- a/opcua/common/utils.py
+++ b/opcua/common/utils.py
@@ -168,7 +168,7 @@ class ThreadLoop(threading.Thread):

     def _create_task(self, future, coro, cb=None):
         #task = self.loop.create_task(coro)
-        task = asyncio.async(coro, loop=self.loop)
+        task = asyncio.ensure_future(coro, loop=self.loop)
         if cb:
             task.add_done_callback(cb)
         future.set_result(task)
```

Please note that while doing `var.set_value(True)` works in this version, by using the alternate version
```
        dv = ua.DataValue(ua.Variant([True], ua.VariantType.Boolean))
        dv.ServerTimestamp = datetime.now()
        dv.SourceTimestamp = datetime.now()
        var.set_data_value(dv) # set node value using implicit data type
```
(with or without ServerTimestamp and/or SourceTimestamp), it yields the same error as the latest version:
```
Traceback (most recent call last):
  File "./client_example_minimal.py", line 31, in <module>
    var.set_data_value(dv) # set node value using implicit data type
  File "/usr/lib/python3.7/site-packages/opcua-0.98.3-py3.7.egg/opcua/common/node.py", line 208, in set_value
  File "/usr/lib/python3.7/site-packages/opcua-0.98.3-py3.7.egg/opcua/common/node.py", line 254, in set_attribute
  File "/usr/lib/python3.7/site-packages/opcua-0.98.3-py3.7.egg/opcua/ua/uatypes.py", line 233, in check
opcua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)
```

I'd like to find the commit that is causing the new code to fail, so I'll try git bisecting the sources. I hope it can be reverted or fixed without having me modify all my Python scripts.
Thank you for your patience.

oroulet, 2018-09-26:
The only recent change is that we do not force ServerTimestamp... It is a performance/memory optimization that any good server should support

fcep, 2018-09-26:
Ok, I'll try to adjust my code to whatever uawrite is doing. Thank you. I'll keep you posted.

fcep, 2018-09-27:
Dear all, I managed to make my code work by using the following:
```
        var = client.get_node("ns=4;s=MAIN.Roof.StopRoofCMD")
        var.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
```
Thank you all for your help.

closed 2018-09-27 by fcep

SudhakarKuma, 2021-04-27:
> Dear all, I managed to make my code work by using the following:
> 
> ```
>         var = client.get_node("ns=4;s=MAIN.Roof.StopRoofCMD")
>         var.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
> ```
> 
> Thank you all for your help.

Hi, I am also working on implementing a client-server model using the OPC-UA protocol. I have a server with two Boolean parameters which I want to modify from a client. The server is as given below:

```Modelica
model randTest
    input Boolean x; 
    Boolean y; 
equation
    y = x; 
end randTest;
```

Now, to change the values, I have a Python OPC-UA client. A part of this client is as given below:

```Python
var = client.get_node("ns=1;i=200000000")
var1 = client.get_node("ns=1;i=250000000")
print(var) 
print(var1)
var.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
var1.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
myVal = client.get_values([client.get_node('ns=1;i=200000000'), client.get_node("ns=1;i=250000000")])
print(myVal)
```

Upon executing the client, the values at the server end are not being changed. Could you please provide some insights into this?

AndreasHeine, 2021-04-27:
#1304

fcep, 2021-04-27:
> Upon executing the client, the values at the server end are not being changed. Could you please provide some insights into this?
> @fcep

Sorry, it's been a long while since I worked on this. I don't remember much, but I don't think I could've helped you anyway. I am by no means an expert on this. Sorry.

AndreasHeine, 2021-04-27:
@SudhakarKuma 
as mentioned in #1304 its look more like an server issue! i tested it with Prosys Simulation Server and Siemens S7-1500.

the way its been written from python-opcua is correct! 
the request to the opc ua server is correct!

as shown in your screenshot:
https://user-images.githubusercontent.com/12606778/115695126-22813800-a37f-11eb-8dfd-ad56fb5a897b.png

it looks like the server does not handle useracesslevel and writemask the right way... if he writes the value with only CurrentRead the server should respond with "BadNotWritable, but he does not... the server just dont return an error... and for sure the server dont change the value!

if you dont believe me than simply download the Prosys Server or Unified Automation Cpp Demo Server or take on of our Example-Servers, and try it hands on. you can also use wireshark to investigate the requests and responses.

please contact the modelica guys for support!

SudhakarKuma, 2021-04-29:
@AndreasHeine Thanks for your explanation. I have also realized that there is some issue with the OPC-UA implementation at the Modelica server side. I will raise this issue with Modelica developers. 

As suggested, I will use Prosys Server to substantiate the point.

PraveenKumarSoni, 2021-07-10:
@oroulet I tried with the above-mentioned solution, but it's not working. Could you please help me??

Here is the error ---
> Requested session timeout to be 3600000ms, got 30000ms instead
ns=3;s=Test_Bulb_Position
Traceback (most recent call last):
  File "/home/read.py", line 32, in <module>
    node_11.set_value(True)
  File "/usr/local/lib/python3.9/dist-packages/opcua/common/node.py", line 217, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/usr/local/lib/python3.9/dist-packages/opcua/common/node.py", line 263, in set_attribute
    result[0].check()
  File "/usr/local/lib/python3.9/dist-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdUnknown: "The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)

And this is the code ---
`   client.connect()

    node_11 = client.get_node("ns=3;s=Test_Bulb_Position") 
    print(node_11)
    node_11.ServerTimestamp = datetime.now()
    node_11.SourceTimestamp = datetime.now()
    node_11.set_value(True)`

Thanks!!

oroulet, 2021-07-10:
If you would read your error message you would know that your node id is wrong.

PraveenKumarSoni, 2021-07-10:
I think this is a correct node id.

oroulet, 2021-07-11:
opcua.ua.uaerrors._auto.BadNodeIdUnknown: "The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)

The node id is 100% sure wrong

AndreasHeine, 2021-07-11:
the error is a response of the opcua server and not an error on clientside you could check the nodeid by browsing to the node with ua expert and then have a look in the attributes tab the first line shows the nodeid of that node! its one of the most common mistakes ...

PraveenKumarSoni, 2021-07-11:
This is the error-

> 

QualifiedName(3:Test_Bulb_Position)
Traceback (most recent call last):
  File "/home/Write.py", line 62, in <module>
    Parking.set_data_value(True)
  File "/usr/local/lib/python3.9/dist-packages/opcua/common/node.py", line 217, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/usr/local/lib/python3.9/dist-packages/opcua/common/node.py", line 263, in set_attribute
    result[0].check()
  File "/usr/local/lib/python3.9/dist-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)'

AndreasHeine, 2021-07-12:
always specify the correct VariantType!

Example for a Double:
```python
        double_node = client.get_node("ns=2;s=Demo.Static.Scalar.Double") # returns a Node-Class
        dv = ua.DataValue(ua.Variant(200.0, ua.VariantType.Double))
        double_node.set_value(dv)
```

PraveenKumarSoni, 2021-07-12:
Okay, I'll try this.

MohammadKurdia, 2021-08-23:
> Dear all, I managed to make my code work by using the following:
> 
> ```
>         var = client.get_node("ns=4;s=MAIN.Roof.StopRoofCMD")
>         var.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
> ```
> 
> Thank you all for your help.

Thanks a lot, this was a great help.

Petro31, 2021-11-18:
Yep, 

> Thanks a lot, this was a great help.

Ditto