FreeOpcUa/python-opcua #709: Unexpected Node in the Server Address Space

bryankim96, 2018-09-25:
I have access to an OPC UA server which serves some nodes. In order to validate it I used a simple example client to examine the nodes, however as I was doing so I ran into a strange error. The test script I used was some very simple boilerplate:

```python
import sys
sys.path.insert(0, "..")
import logging
import time

from opcua import Client, ua

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    client = Client(SERVER_IP)
    try:
        client.connect()
        client.load_type_definitions()  

        root = client.get_root_node()
        objects = client.get_objects_node()

        for child in objects.get_children():
            print("*********")
            print(child)
            print(child.nodeid)
            print("*********")

    finally:
        client.disconnect()
```

I was able to connect to the server without any issues, examine some nodes, etc.. However, I then discovered something very strange. When I examined the objects folder using a different OPC UA client (the Prosys OPC UA client), I found 6 nodes as immediate children of the object folder (which was what I was expecting based on my knowledge of the server design). However, when I ran the above client script, it printed 7 nodes, one of which did not correspond to any of the expected nodes (I matched them up by nodeid). This unexpected extra node had a nodeid of StringNodeId(ns=2;s=None). However, when I try to retrieve any attributes of this node, for example using .get_browse_name(), it fails with:

`opcua.ua.uaerrors._auto.BadNodeIdUnknown: The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)`

because it is unable to locate that node.

Do you know what the cause/meaning of this could be?

Also, on an unrelated topic, is there any way to list/retrieve all attributes of a particular node? I noticed that Node.get_attributes(attrs) supports retrieving multiple attributes but requires that you know them in advance and pass them in as arguments.

bryankim96, 2018-09-26:
> interesting... you will need to record a session with wireshark so we understand what the server is sending...

Ok, I captured the ReadRequests and ReadResponses with Wireshark. I took a look at the records but I'm not sure if there's any useful information for diagnosing the problem. What sort of information is useful for me to provide to you? Should I just send you a capture file?

> to read all attributed maybe you can do something like that

Perfect, thanks!

oroulet, 2018-09-26:
Yes attach to bug report

bryankim96, 2018-09-26:
Attached is the pcapng file containing only the packets related to the OPC UA communication.

I slightly simplified the above client script to the following to examine only the extra node:

```python

import sys
sys.path.insert(0, "..")
import logging
import time

from opcua import Client, ua

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    client = Client(SERVER_IP)
    try:
        client.connect()
        client.load_type_definitions()

        root = client.get_root_node()
        objects = client.get_objects_node()

        node = client.get_node('ns=2;s=None')

        print(node)
        print(node.nodeid)
        try:
            print(node.get_browse_name())
        except:
            pass

    finally:
        client.disconnect()
```

[opcua_capture.zip](https://github.com/FreeOpcUa/python-opcua/files/2420657/opcua_capture.zip)

oroulet, 2018-09-26:
Sorry we need a session record from browsing the parent. I want to know why
we get such a bad node as child

bryankim96, 2018-09-26:
The parent node is the objects node, which I do return with client.get_objects_node(). Is there something more specific you need from the objects node?

oroulet, 2018-09-26:
I want to see the record from server answering get_children().  Btw which
version? Can you updatefirst?

bryankim96, 2018-09-27:
I'm using the most recent version (0.98.5), installed via pip.

Here is the attached wireshark capture from running the following script:

```python

import sys
sys.path.insert(0, "..")
import logging
import time

from opcua import Client, ua

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    client = Client(SERVER_IP)
    try:
        client.connect()
        client.load_type_definitions()

        root = client.get_root_node()
        objects = client.get_objects_node()

        print(root.get_children())
        print(objects.get_children())
        
    finally:
        client.disconnect()

```

The printed output was:

`[Node(TwoByteNodeId(i=87)), Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86))]
[Node(FourByteNodeId(i=2253)), Node(StringNodeId(ns=2;s=None)), Node(StringNodeId(ns=2;s=MPESFolder)), Node(StringNodeId(ns=2;s=ACTFolder)), Node(StringNodeId(ns=2;s=PSDFolder)), Node(StringNodeId(ns=2;s=UnknownMirror)), Node(StringNodeId(ns=2;s=Panel_0))]`

[opcua_capture_2.pcapng.gz](https://github.com/FreeOpcUa/python-opcua/files/2425416/opcua_capture_2.pcapng.gz)

oroulet, 2018-09-27:
yes this the correct code but I need the complete log! you only sent 3 unrelated read operations.  send me everything

bryankim96, 2018-09-27:
Sorry, here is the full unfiltered log:

[opcua_capture_3.pcapng.gz](https://github.com/FreeOpcUa/python-opcua/files/2425493/opcua_capture_3.pcapng.gz)

oroulet, 2018-09-27:
[image]

wireshark says that there is 5 nodes with the second one being with Null string as NodeID......either wireshark also is wrong on there is something wrong with your server. What server and how were this nodes created...

bryankim96, 2018-09-27:
This sounds right to me, in the second output (objects.get_children()), I got 7 nodes printed. The only thing that I don't understand is what that 2nd node with the Null string NodeID means...

The server was written with the Unified Automation C++ SDK and the nodes were added to the server in the standard way using addNodeAndReference().

oroulet, 2018-09-28:
For now it looks like the error is in server.....

oroulet, 2018-09-28:
the prosys client probably filter that node because it is invalid while we just return everything. I think that returing it to users is the correct thing to do so users see the error and can fix the server or make a bypass on client side

bryankim96, 2018-09-28:
I see... I'm not sure why or how this would happen, but I'll look into the server and see if I can find anything suspicious.

Thanks.

bryankim96, 2018-10-02:
With some help I was able to find out the cause of the error. As you suspected, it was due to a bug in the server-side code. Thanks for the help.

closed 2018-10-02 by bryankim96