FreeOpcUa/python-opcua #710: UA Expert not being able to connect to Server

JFKane, 2018-09-25:
Hi all,

I want to set a Beaglebone Black as a server sending sensor data (right now it's just random numbers) to my laptop over OPC UA and reading it with UA Expert (OPC Client)

Here is my server.py 
`
from opcua import ua,Server
from random import randint
import time
import datetime

server = Server()

url = "ocp.tcp://128.130.39.50:4840"
server.set_endpoint(url)
server.set_security_policy([
	ua.SecurityPolicyType.NoSecurity,
	ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
	ua.SecurityPolicyType.Basic128Rsa15_Sign,
	ua.SecurityPolicyType.Basic256_SignAndEncrypt,
	ua.SecurityPolicyType.Basic256_Sign
])
name="OPCUA_BB_Server"
addspace = server.register_namespace(name)

node = server.get_objects_node()

Param = node.add_object(addspace, "Parameters")

Temp = Param.add_variable(addspace, "Temperature", 0)
Temp.set_writable()

server.start()
print("Server started at {}".format(url))

while True:
	Temperature = randint(0, 100)
	print(Temperature)
	Temp.set_value(Temperature)
	
	time.sleep(2)
`

However, everytime I try to connect UA Expert to the server I get 
`
The server does not support the configured security policy 'http://opcfoundation.org/UA/SecurityPolicy#None'
`

I've tried enabling different configurations on UA Expert (Basic128, Basic256 etc.) but always the same error. Any ideas what could be the problem?

ibrahimsadek, 2018-09-26:
I am not sure, but if you select one security option in the server and client side let's say:

server.set_security_policy([ua.SecurityPolicyType.Basic256_SignAndEncrypt]), then the uaExpert as a client should respond properly.

JFKane, 2018-09-26:
Thank you for the suggestions, I tried all scenarios (no security policy, changing UA Expert configuration etc.) but it always returns:

```
18:04:52.416 | Server Node        |                                | Connecting failed with error 'BadConfigurationError'
18:04:52.401 | Server Node        |                                | Error 'BadConfigurationError' was returned during ActivateSession
18:04:52.358 | Server Node        |                                | The server returned no certificate, all certificate checks will be skipped.
18:04:52.354 | Server Node        |                                | Used UserTokenType: Anonymous
18:04:48.936 | Server Node        |                                | The server does not support the configured security policy 'http://opcfoundation.org/UA/SecurityPolicy#None'.
18:04:48.936 | Server Node        |                                | ApplicationUri: ''
```

I'm guessing the problem is not the configuration itself but somewhere else, unfortunately I'm kinda stuck...

startpyy, 2019-07-07:
I got the same Issue. 
Did you find any solution? =)

AndreasHeine, 2019-11-11:
security_policy =   [
                            ua.SecurityPolicyType.NoSecurity,
                            ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
                            ua.SecurityPolicyType.Basic256Sha256_Sign
                        ]
    server.set_security_policy(security_policy)
    policyIDs =   [
                    "Anonymous", "Basic256Sha256", "Username"
                ]
    server.set_security_IDs(policyIDs)