FreeOpcUa/python-opcua #711: Cannot read single array elements

JJdeVries, 2018-10-01:
Hi,

I'm trying to read/write single array elements from an OPCUA server, but I don't seem to be able to. I'm running python-opcua version 0.98.5, and have a minimum client/server example:

**The server:**
```
import time
import logging

from opcua import ua, Server

if __name__ == "__main__":
    logging.basicConfig()
    # Setup our server
    server = Server()
    server.set_security_policy([ua.SecurityPolicyType.NoSecurity])
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")
    server.set_server_name("Minimal Array Example")

    # setup our own namespace
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # populating our address space
    myvar = server.nodes.objects.add_variable(idx, 'test_array', [6.7, 7.7, 23, 14, 5])

    # starting!
    logging.info("Starting the server!")
    server.start()
    try:
        while True:
            time.sleep(0.1)
    except KeyboardInterrupt:
        pass
    finally:
        server.stop()
```

**The client:**
```
import logging

from opcua import Client, ua

def get_array_elems(node, index_range):
    # Construct a ReadValueId object:
    rv = ua.ReadValueId()
    rv.NodeId = node.nodeid
    rv.AttributeId = ua.AttributeIds.Value
    rv.IndexRange = index_range

    # Construct the ReadParameters object
    params = ua.ReadParameters()
    params.NodesToRead.append(rv)

    # Read the data from the server.
    result = node.server.read(params)
    result[0].StatusCode.check()
    return result[0].Value.Value

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    client = Client("opc.tcp://localhost:4840/freeopcua/server/")
    try:
        client.connect()
        root = client.get_root_node()
        test_var = root.get_child(["0:Objects", "2:test_array"])

        # If we simply get the value we expect an array of size 5
        val = test_var.get_value()
        if len(val) != 5:
            logging.error("Unexpected length of test_var: {} != 5".format(len(val)))

        # But get_array_dimensions returns an empty list?
        array_dim = test_var.get_array_dimensions()
        logging.error("Array dimensions is {}".format(array_dim))

        # Reading a single array element does not behave as expected.
        single_elem = get_array_elems(test_var, "1")
        logging.error("Single elem: {}".format(single_elem))

        # Also reading part of an array simply gives back the entire array.
        part_array = get_array_elems(test_var, "2:4")
        logging.error("Part of array {}".format(part_array))

    finally:
        client.disconnect()
```
As expected the client example's `test_var.get_value()` returns the correct array of size five. The `get_array_elems` method now mimicks the implementation of the node class `get_attribute` while filling in the `IndexRange` parameter with the desired index range. I would now expect the returned values to be some subset of the test_array, but instead the complete array is always returned.

Have I not initialized the `test_array` correct? (seeing that also `get_array_dimensions` returns an empty list), or is this functionality not supported?

oroulet, 2018-10-02:
It might be that the index range stuff is not implemented... Try to look at code somewhere in address_space.py...

JJdeVries, 2018-10-02:
@oroulet Ahh yes you are right, `IndexRange` is used nowhere in the `AttributeService` class, or `get_attribute_value`. Adding the IndexRange to reading should not be too difficult, as that could be handled in the AttributeService I think, but I'll probably also need the element specific write, which will be more work.

I'm not sure if I will have to time to solve the issues, but if i'll manage i'll post a pull request

GrantEdwards, 2020-05-06:
I need to be able to write to a range of bytes (or just a single byte) within a node that is a byte array.  Is this still something that's not supported?