FreeOpcUa/python-opcua #714: Client timeout does not work for initial socket connection

Obihoernchen, 2018-10-08:
The timeout you can set for the Client doesn't work for the initial socket connection.

To recreate this just set an IP of a network you can't reach. For instance I used `10.0.0.1`.
```python
from opcua import Client

with Client(url='opc.tcp://10.0.0.1', timeout=1):
    print("successfully connected")
```
It will eventually time out, but not after 1 second. It depends on the OS and OS network timeout settings.
It usually takes several seconds or minutes.
```
Traceback (most recent call last):
  File "without_timeout.py", line 3, in <module>
    with Client(url='opc.tcp://10.0.0.1', timeout=1):
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 126, in __enter__
    self.connect()
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 256, in connect
    self.connect_socket()
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 281, in connect_socket
    self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 226, in connect_socket
    return self._uasocket.connect_socket(host, port)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 139, in connect_socket
    sock = socket.create_connection((host, port))
  File "/usr/lib64/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib64/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
```

The issue is, that the socket is created without a timeou in: https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/ua_client.py#L139
I patched `UaSocketClient.connect_socket()` and added a timeout to the socket.
```python
import socket

from opcua import Client, ua
from opcua.client.ua_client import UASocketClient

# patch UASocketClient.connect_socket to force a proper socket timeout for initial connection. Afterwards the timeout is set to None again.
def connect_socket(self, host, port):
    """
    connect to server socket and start receiving thread
    """
    self.logger.info("opening connection")
    # Create socket with timeout for initial connection
    sock = socket.create_connection((host, port), timeout=self.timeout)
    # set to blocking mode again
    sock.settimeout(None)
    # nodelay ncessary to avoid packing in one frame, some servers do not like it
    sock.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
    self._socket = ua.utils.SocketWrapper(sock)
    self.start()

# replace the function
UASocketClient.connect_socket = connect_socket

with Client(url='opc.tcp://10.0.0.1', timeout=1):
    print("successfully connected")
```

Now I get a proper timeout after 1 second.
```
Traceback (most recent call last):
  File "with_timeout.py", line 24, in <module>
    with Client(url='opc.tcp://10.0.0.1', timeout=1):
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 126, in __enter__
    self.connect()
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 256, in connect
    self.connect_socket()
  File "/usr/local/lib/python3.6/site-packages/opcua/client/client.py", line 281, in connect_socket
    self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 226, in connect_socket
    return self._uasocket.connect_socket(host, port)
  File "with_timeout.py", line 13, in connect_socket
    sock = socket.create_connection((host, port), timeout=self.timeout)
  File "/usr/lib64/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib64/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
socket.timeout: timed out
```

Is it by design, that there is no timeout for the initial connection? This confused me, because I thought I set a timeout, but it didn't timeout in time. If you think the initial socket creation should use the timeout you pass to the client, I could submit a PR.

oroulet, 2018-10-13:
There is not much doc in Python opcua but this variable is documented as session timeout in doc string ;-) . As far as remember you can set the socket timeout on the UaServer object... But I may be wrong and if you manage to get overview over all timeouts in this library, and this one is missing, then a PR is welcome

Obihoernchen, 2018-10-14:
Mhh UaServer is just the example server, not the client.

`opcua.client.ua_client UASocketClient` has a timeout setting and this is set [here](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/ua_client.py#L26). It gets passed from the high level Client to `UaClient` and then to `UaSocketClient`. It's used in all requests but not for the initial socket creation [here](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/ua_client.py#L139). I can't find any other socket timeout.

I found an older PR: https://github.com/FreeOpcUa/python-opcua/pull/316 which added a seperate socket timeout, but kept this timeout afterwards, which caused issues, so it got reverted in: https://github.com/FreeOpcUa/python-opcua/commit/cb034f5f44da55fba7291318258f67e61c32f81e
That's why I've set `sock.settimeout(None)` after the initial connection again.

I would propose to change the following in [`opcua.client.ua_client UASocketClient.connect_socket()`](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/ua_client.py#L139).
Instead of:
```python
self.logger.info("opening connection")
sock = socket.create_connection((host, port))
```
do
```python
sock = socket.create_connection((host, port), timeout=self.timeout)
# set to blocking mode again
sock.settimeout(None)
```
This should fix long timeouts for the initial socket creation, but after the initial connection it should work as it does right now (socket with no timeout).

I can submit a PR, but first of all I would like to hear your opinion on this.

oroulet, 2018-12-12:
@Obihoernchen @Henna1977 Sorry I completely forgot that one and thought it was already merged. yes it sound appropriate and PR is welcome

commit 77263b85 (2019-01-15): Add timeout to initial socket creation (fix #714)
closed 2019-01-15 by oroulet