FreeOpcUa/python-opcua #715: server.import_xml leading to BadParentNodeIdInvalid or BadNodeIdExists

NNTin, 2018-10-09:
I took the [server-minmal.py example](https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-minimal.py) and added the following line between line 12 and 13:

`server.import_xml("absolute-path-of-my-xml-file.xml")`

It works when importing [Opc.Ua.Di.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/master/DI/Opc.Ua.Di.NodeSet2.xml).  
But when I import [Opc.Ua.Fdi5.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/master/FDI/Opc.Ua.Fdi5.NodeSet2.xml) I am getting the following error:

```
failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=2))
Traceback (most recent call last):
  File "Z:/Tin/python/experiment/server.py", line 13, in <module>
    server.import_xml("Z:/prodisys/Tin/python/xml/Opc.Ua.Fdi5.NodeSet2.xml")
  File "Z:\python-opcua\opcua\server\server.py", line 523, in import_xml
    return importer.import_xml(path, xmlstring)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 88, in _add_node_data
    node = self.add_variable_type(nodedata)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 340, in add_variable_type
    res[0].StatusCode.check()
  File "Z:\python-opcua\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)
```

When I remove the troubling node from the xml the error changes to:  
`failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=2))`

When removing that again it changes to:
`failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=92))`

and so on.

When I import [Opc.Ua.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/master/Schema/Opc.Ua.NodeSet2.xml) I am getting the following error:

```
AddNodesItem: Requested NodeId NumericNodeId(i=3062) already exists
failure adding node NodeData(nodeid:NumericNodeId(i=3062))
Traceback (most recent call last):
  File "Z:/Tin/python/experiment/server.py", line 13, in <module>
    server.import_xml("Z:/prodisys/Tin/python/xml/Opc.Ua.NodeSet2.xml")
  File "Z:\python-opcua\opcua\server\server.py", line 523, in import_xml
    return importer.import_xml(path, xmlstring)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 82, in _add_node_data
    node = self.add_object(nodedata)
  File "Z:\python-opcua\opcua\common\xmlimporter.py", line 183, in add_object
    res[0].StatusCode.check()
  File "Z:\python-opcua\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)
```

I've been trying to figure out how I can import those xml files. I'm glad over any help.

oroulet, 2018-10-13:
Not being able to import NodeSet2 is completely normal. It is already imported. Concerning the other one either it is a bug or it requires another nodeset before

NNTin, 2018-10-19:
Thanks for clarifying regarding NodeSet2.

How would I import [Opc.Ua.Fdi5.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/master/FDI/Opc.Ua.Fdi5.NodeSet2.xml)?

In SiOME I would first import [Opc.Ua.Di.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/master/DI/Opc.Ua.Di.NodeSet2.xml) and then Opc.Ua.Fdi5.NodeSet2.xml.

When I do:

```
self.server.import_xml("Z:/prodisys/Tin/python/xml/Opc.Ua.Di.NodeSet2.xml")
self.server.import_xml("Z:/prodisys/Tin/python/xml/Opc.Ua.Fdi5.NodeSet2.xml")
```
I'm getting: `The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)`

Di.NodeSet2 is built on top of NodeSet2. Fdi5.Nodeset2 is built on top of Di.NodeSet2.

Is there a different method for importing multiple xmls?

oroulet, 2018-10-19:
the code above is correct, might be a bug in importer.... you are welcome to try to find out. Unfortumately the code is hard to understand...If someone has ideas to improve it , it would be great...

PrudhviRajC, 2019-02-24:
Hallo @NNTin, 
I am implementing the OPC UA Server and for that I want to import the XML file into it. When I import, I am also facing the same issue -
"raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)". 

Did you find any solution for this so far?
Thank you

oroulet, 2019-02-24:
BadNodeExist is clearly an issue with your setup or the XML file itself, not  python-opcua code

NNTin, 2019-02-24:
My error code was different. The solution to my error was to import both models into SiOME and then export it as a single xml file.

I don't know if this solves your problem. Best of luck!

PrudhviRajC, 2019-02-24:
Sorry, I pasted the message wrong.
Its 
"raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)"

[image]

oroulet, 2019-02-27:
I just found and fixed two bugs in apcua-asyncio. they should be backported to python-opcua

https://github.com/FreeOpcUa/opcua-asyncio/commit/4b85a3f12242a59faf2cec01150e2124e7875d37
and
https://github.com/FreeOpcUa/opcua-asyncio/commit/681bcd6230d936aa838a44f7eee09d76eef1713b

Can someone do it?

oroulet, 2019-02-27:
BUT we now add a default reference type when no reference type is specified. I have no idea what the defalt should be. I used HasComponent, it might be wrong...

biancode, 2019-04-27:
models/Opc.Ua.Di.NodeSet2.xml does not work for my on import XML see: 

> INFO:opcua.common.xmlimporter:Importing xml node (QualifiedName(1:IVendorNameplateType), NumericNodeId(ns=1;i=15036)) as (QualifiedName(2:IVendorNameplateType) NumericNodeId(ns=2;i=15036))
> DEBUG:opcua.server.address_space:Adding node NumericNodeId(ns=2;i=15036) QualifiedName(2:IVendorNameplateType)
> INFO:opcua.server.address_space:add_node: while adding node NumericNodeId(ns=2;i=15036), requested parent node NumericNodeId(i=17602) does not exists
> WARNING:opcua.common.xmlimporter:failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=15036))
> Traceback (most recent call last):
>   File "/home/biancode/PycharmProjects/python-opcua-server/server-with-encryption.py", line 69, in <module>
>     server.import_xml("./models/Opc.Ua.Di.NodeSet2.xml")
>   File "/home/biancode/.local/lib/python3.6/site-packages/opcua/server/server.py", line 463, in import_xml
>     return importer.import_xml(path, xmlstring)
>   File "/home/biancode/.local/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 67, in import_xml
>     node = self._add_node_data(nodedata)
>   File "/home/biancode/.local/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 84, in _add_node_data
>     node = self.add_object_type(nodedata)
>   File "/home/biancode/.local/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 196, in add_object_type
>     res[0].StatusCode.check()
>   File "/home/biancode/.local/lib/python3.6/site-packages/opcua/ua/uatypes.py", line 233, in check
>     raise UaStatusCodeError(self.value)
> opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)

v0.98.6 I mean

anshi43, 2023-06-05:
Hello guys, 

I am also getting the same error as yours : **raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)**

i am trying to import all these files. custom_nodeset.xml was created using siemens SiOME modelling editor. here is the code : 
# import opcua 
# import time
# from opcua import Server
# import random

# class OPCServer:
#      def __init__(self) :
#          self.server = Server()
#      def start(self):
#          self.server.set_server_name("OPC UA Server")
#          self.server.set_endpoint("opc.tcp://192.168.0.107:4140")
#         
         
#          #self.server.import_xml("Opc.Ua.Di.Nodeset2.xml")
#          #self.server.import_xml("Opc.Ua.Machinery.Nodeset2.xml")
#          #self.server.import_xml("Opc.Ua.Pumps.Nodeset2.xml")
#          self.server.import_xml("custom_nodeset.xml")
#          self.server.start()
        
         
            
# if __name__ == '__main__':
#      server = OPCServer()
#      server.start()
#       #server.run()

How did you guys solve this error? 
your comments and advice would be appreciated.

anshi43, 2023-06-05:
> See #1475 . Switch to asyncua

i tried but the error remain same

anshi43, 2023-06-06:
> Sorry, I pasted the message wrong. Its "raise UaStatusCodeError(self.value) opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)"
> 
> [image]

did you find the solution because i am getting the same error ?

AndreasHeine, 2023-06-06:
the error tells you that the parent already exist! the nodeset ia already part of the default addreassspace.