FreeOpcUa/python-opcua #716: Question: can you write SourceTimestamp

betrri, 2018-10-10:
Hello,

I'm about to do some testing and like to know if I'm able to write custom datetime value to DataValue.SourceTimestamp. Right now it feels like it's not possible. The code is as follows:

test = str(ns + "s=" + name)
		var = self.client.get_node(test)
		dv = var.get_data_value()
		dv.SourceTimestamp = datetime.datetime(int(timestamp[0]).......

timestamp contains datetime values (year, month etc..) in a list

Best Regards 
Petri

oroulet, 2018-10-13:
Your code above does not even try to write something ..

betrri, 2018-11-22:
Sorry I forgot to add the set_value command. This just strangely sets the SourceTimestamp to 1999-11-30 00:00:00. Any idea if it's even possible to write the SourceTimestamp to a pre-defined value?

	def opcua_write_timestamp(self, name, timestamp, ns="ns=1;"):
		test = str(ns + "s=" + name)
		var = self.client.get_node(test)
		dv = var.get_data_value()
		dv.SourceTimestamp = datetime.datetime(int(timestamp[0]),int(timestamp[1]),int(timestamp[2]),int(timestamp[3]),int(timestamp[4]),int(timestamp[5]))
		var.set_value(dv)