FreeOpcUa/python-opcua #720: Get Session number on client site

RasPiTool, 2018-10-26:
Hello folks,

I can on server site use expression like this
`server.bserver.clients`
to get information about connected clients. In the info-list I can see the IP >192.168.0.103< and Session >5184< number (I think it is the Session number).
`[OPCUAProtocol(('192.168.0.103', 5184), InternalSession(name:('192.168.0.103', 5184), user:User.Anonymous, id:NumericNodeId(i=11), auth_token:NumericNodeId(i=1001)))]`
How can I get this information on client site. It's easy to get the client IP but how can I get the Session number?

RasPiTool, 2018-10-26:
Unfortunately not. The idea is to send multiple simulations to the server and get a real-time solution. I want to have more flexibility. For example.
1. The session is running and I get a solution in real time.
2. Now I have a second simulation and want to simulate it with the server. The server needs to know what the first and second sessions are to control the variable, and so on.

zerox1212, 2018-10-26:
I don't really understand why you need the session id for this. Why not just have each client send it's own ID?

RasPiTool, 2018-10-26:
The problem is when multiple sessions are started at the same time. And the sessions send it's own ID on regular node.

oroulet, 2018-10-30:
@RasPiTool when calling connect() on the client side, we create a secure connection then a session. If you do not use connect(),  you can look at code, and do everything by hand, then you can get the entire session reply from the server and I am sure there is a session ID in there

RasPiTool, 2018-11-01:
thanks for answer,
I use client.connect().
Here is my example code.

```
client = Client("opc.tcp://admin@" + server_ip + ":4840/freeopcua/server/")
client.connect()
se = client.create_session()
print(se)
```
My Output:
```
CreateSessionResult(SessionId:NumericNodeId(i=47), AuthenticationToken:NumericNodeId(i=1037), RevisedSessionTimeout:3600000.0, ServerNonce:b'\xe3\xe5\xfd;csU\x87\xba\xf2\n\xf1"\x9bQ\x8c2\xaeD\xc82Q\xd6E\x12\x02\xecN\xf3%\x82Y', ServerCertificate:None, ServerEndpoints:[EndpointDescription(EndpointUrl:opc.tcp://192.168.0.119:4840/freeopcua/server/, Server:ApplicationDescription(ApplicationUri:urn:freeopcua:python:server, ProductUri:urn:freeopcua.github.io:python:server, ApplicationName:LocalizedText(Encoding:2, Locale:None, Text:FreeOpcUa Example Server), ApplicationType:ApplicationType.ClientAndServer, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryUrls:['opc.tcp://0.0.0.0:4840/freeopcua/server/']), ServerCertificate:None, SecurityMode:MessageSecurityMode.None_, SecurityPolicyUri:http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens:[UserTokenPolicy(PolicyId:anonymous, TokenType:UserTokenType.Anonymous, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:None), UserTokenPolicy(PolicyId:certificate_basic256, TokenType:UserTokenType.Certificate, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:None), UserTokenPolicy(PolicyId:certificate_basic128, TokenType:UserTokenType.Certificate, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:None), UserTokenPolicy(PolicyId:username, TokenType:UserTokenType.UserName, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:None)], TransportProfileUri:http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel:0)], ServerSoftwareCertificates:[], ServerSignature:SignatureData(Algorithm:http://www.w3.org/2000/09/xmldsig#rsa-sha1, Signature:b''), MaxRequestMessageSize:65536)
```
On server site I get with method 
```
server.bserver.clients
```

This output:

```
OPCUAProtocol(('192.168.0.103', 45310), InternalSession(name:('192.168.0.103', 45310), user:User.Anonymous, id:NumericNodeId(i=47), auth_token:NumericNodeId(i=1037)))]

```
That's the same on both _sides NumericNodeId(i=47) NumericNodeId(i=1037)_
Can you explain me what is  id:NumericNodeId(i=47), auth_token:NumericNodeId(i=1037) ? 
Sorry, but I am new in programming and opcua.
I found the node i=47 in the server.
[image]
If disconnect and connect to server again I get _NumericNodeId(i=23) NumericNodeId(i=1030)_
And this Node I found here:
[image]

I did not find the node i = 1037 and 1030. Why the nodes i=47 and i=23 are on different places?

Thank you in advance